arduino/Arduino #3781: Serial Monitor window not appearing

Josephc3, 2015-09-11:
I have Arduino software version 1.6.5  and the project where the problem began was at the time programming an arduino uno.
Accidentally pressed the keyboard shortcut to open the serial monitor as the code was uploading and the status window came back with an error that the port could not be opened. I then re uploaded the code, which was confirmed by the correct operation of the hardware.

Since then, opening the serial monitor results in a new tab appearing in windows (windows 10 home), however no visible window. Tried keyboard shortcuts to move windows and it isnt hidden anywhere, its just not there.

Reset the software and PC to no avail. Tried different sketches on the arduino, which upload correctly. Also the serial port on the PC is fine as i can establish communications through RealTerm.

Josephc3, 2015-09-11:
Resolved by setting last.serial.location=200,200,200,200 in preferences.txt, unsure as to what created the issue to begin with but these values were set very high before i changed them suggesting that the serial monitor was pushed off my display by quite a long way

VogonJeltz, 2015-09-11:
congrats for your home run! :)

closed 2015-09-14 by ffissore

robo-shark, 2016-10-31:
lowering screen resolution from 1920x1080 to 1366x768 then running serial monitor made it appear, then I just changed back the resolution to the way it was and everything was OK

jschmuland, 2018-02-03:
I tried changing the preference.txt to no avail however what worked for me (If anyone still reads this thread like I just did) was to: 
hover over the tab on the task bar,
right click on the preview,
select maximize window,
then I just dragged it into the middle of my screen.

That was for windows. If you are on a mac, I guess you just have to figure out how to maximize the window yourself.

NicksonYap, 2018-11-29:
@jschmuland That works!

I suppose this happens to multi-screen devices

EliteArmedForce, 2019-03-17:
@jschmuland Thank you, brother.  It worked with me.
@NicksonYap I noticed that.

ben-student, 2020-06-13:
@NicksonYap Thanks dude , finally worked.