arduino/Arduino #3782: RCTime tutorial on website is corrupted--needs fixing

ElectricRCAircraftGuy, 2015-09-11:
Something happened to it. The tutorial code is scrambled and malformatted. 

https://www.arduino.cc/en/Tutorial/RCtime

partial screenshot:

[image]

carise, 2015-09-12:
Poking around on the Chrome Dev Tools, it seems that if I uncheck the "white-space: normal" css property on the pre tag, the formatting goes back.

[image]

agdl, 2015-09-14:
@ElectricRCAircraftGuy done it is ok now. I'm closing this!

closed 2015-09-14 by agdl