arduino/Arduino #3783: Compiler warnings when compiling EEPROM.h

nickgammon, 2015-09-12:
Under 1.6.4 of the IDE (and reportedly under nightly builds as well) compiling an EEPROM example gives this warning:

```
In file included from eeprom_get.ino:18:0:
/home/nick/Development/arduino-1.6.4/hardware/arduino/avr/libraries/EEPROM/EEPROM.h:43:30: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     operator const uint8_t() const       { return **this; }
                              ^
/home/nick/Development/arduino-1.6.4/hardware/arduino/avr/libraries/EEPROM/EEPROM.h:92:26: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     operator const int() const          { return index; }
                          ^
```

To fix, remove the **first** const from both of those lines.

Chris--A, 2015-09-12:
Cheers, in the update proposal I have in the works, all the warnings are fixed.

Once all the changes are pushed, I'll link here and add a post the dev mailing list. :+1:

maxmclau, 2016-02-25:
Any update on this?

Chris--A, 2016-02-25:
Yeah, I have the updated files ready, I'll do up a PR after work tomorrow.

glansberry, 2017-01-12:
Bump - problem is still there in 1.8.0

Chris--A, 2017-01-13:
This has been fixed and added to the next proposal (https://github.com/Chris--A/Arduino/commit/acbd8bccdc13b14098adff0330403dbf010ed430) [here](https://github.com/arduino/Arduino/compare/master...Chris--A:EEPROM_2.1?expand=1)

per1234, 2017-07-04:
@Chris--A I can't seem to find the pull request.

gtozzi, 2017-12-07:
It looks like this is still present in 1.8.5

szotsaki, 2018-04-10:
Any news on removing the two `const` qualifiers?

d-a-v, 2018-04-11:
Do you think making a PR would help with this 2½ years old trivial fix ?

nickgammon, 2018-04-11:
It can't make things any slower. :)

matthijskooijman, 2018-08-30:
I'm closing this in favor of https://github.com/arduino/ArduinoCore-avr/issues/36 which describes the same issue, but is in the (by now) right repository (and has a PR with a fix).

closed 2018-08-30 by matthijskooijman