arduino/Arduino #3786: [DOC] https://www.arduino.cc/en/Serial/IfSerial only mentions Leonardo

q2dg, 2015-09-12:
It should changed by "32U4-based boards or something like this.

Moreover...what about Zero or Due? The example piece...

     while (!Serial) {
         ; // wait for serial port to connect. Needed for Leonardo only
       }

is needed for them too??

q2dg, 2015-09-12:
Well...in https://www.arduino.cc/en/Main/ArduinoBoardZero there is a note saying: "In contrast to some Arduino boards (e.g. Uno), when you open the serial monitor and the board is connected through the Programming Port the board does not automatically reset." So I deduce with Zero it is necessary to write the piece *while(!Serial) {;}* ...using both (Programming and Native) USB ports??? It isn't clear.

In Due I suppose when using Programming USB Port with Serial Monitor board is reset because uses the same 16U2 chip than UNO, but when using Native USB Port it isn't...but it isn't clear, again.

matthijskooijman, 2015-09-12:
Yup, the Zero and Due behave the same on the native USB port AFAIK, so that example would probably need to be updated to something more general (like "Needed for boards without auto-reset only").

q2dg, 2015-09-12:
So my question is: which boards haven't auto-reset? My guess is:
* 32U4 based: Micro, Yún, LilypadUSB
* Native USB Port in Due
* **Both** USB Ports in Zero (as the note in its product's page says)

I think a paragraph like above (corrected if needed) should be added to https://www.arduino.cc/en/Serial/IfSerial

agdl, 2015-09-14:
On the Zero no way to use while(!Serial) with the programming because the serial is always open and connected to the EDBG. The sentence has to be changed to needed for native USB boards only. I will make a PR for all the examples ASAP

q2dg, 2015-09-21:
So...I infer serial monitor doesn't autoreset the Zero neither using programming nor native usb port, does it? I haven't any to try it.

cmaglie, 2015-09-21:
@q2dg 
Yes, correct.

agdl, 2015-09-21:
here the PR https://github.com/arduino/Arduino/pull/3833

Talking with @cmaglie we chose the phrase:

wait for serial port to connect. Needed for native USB port only

q2dg, 2015-09-21:
Well, I don't wanna bother you more than necessary, but I would specify I little more...something like this

"wait for serial port to connect. Needed for native USB port only (**i.e: ATmega32U4-based boards' only port and Due's one)**"

A newbie doesn't need to know what is a "native" USB port, don't you think?

agdl, 2015-09-22:
It is too long as a comment, I think it's better to improve the online documentation. In this way if someone has doubts he can easily search for an aswer

q2dg, 2015-09-22:
Ok, so waiting for this improvement in online documentation, I close this issue, if you don't mind.
Thanks!!

closed 2015-09-22 by q2dg