arduino/Arduino #3787: Arduino website code corrupted on multiple web pages

ElectricRCAircraftGuy, 2015-09-12:
See here: https://github.com/arduino/Arduino/issues/3782

Here's two more pages with corrupted displaying of example code.
1) https://www.arduino.cc/en/Tutorial/ShiftIn
2) https://www.arduino.cc/en/Tutorial/ShftIn21

On a sidenote, please add RCtime, ShiftIn, and ShiftOut (https://www.arduino.cc/en/Reference/ShiftOut) Tutorials to the core Arduino examples here: https://www.arduino.cc/en/Tutorial/BuiltInExamples. They are very helpful, but a bit tricky to find on the Arduino website currently.

ElectricRCAircraftGuy, 2015-09-13:
Here's another one: http://playground.arduino.cc/Code/CapacitiveSensor

-I think it should be added to the main Arduino examples too.

per1234, 2015-09-14:
>Here's another one with code with messed up formatting: http://playground.arduino.cc/Code/CapacitiveSensor

I was able to fix that one because it's in the playground.

agdl, 2015-09-14:
@ElectricRCAircraftGuy @per1234 I fixed them. For what concern adding them to the core please if you want to do so open a pull request in which you add them and it will be evaluated.

If it is ok for you close this issue please.

ElectricRCAircraftGuy, 2015-09-14:
@agdl Arturo, sounds good. It looks like I edit core examples here (in my forked version, of course), right? https://github.com/arduino/Arduino/tree/bd8f7932e64fb3a41e3d934437d5eabd8707dcc9/build/shared/examples

As for the web-pages that go with each example, though (), how are those created?
https://www.arduino.cc/en/Tutorial/BuiltInExamples

ElectricRCAircraftGuy, 2015-09-14:
@per1234 , thanks for fixing the Playground one. I notice it looks a little different than this page's code, for instance, however (https://www.arduino.cc/en/Tutorial/ShftIn21).

The playground code looks double-spaced, which is kind of odd, and it doesn't have the nice syntax highlighting like the Shftln21 example above has. Any idea how to make it look more like the code linked above?

@agdl, perhaps you can add some insight here?

per1234, 2015-09-14:
> I notice it looks a little different than this page's code

@ElectricRCAircraftGuy the Arduino Playground uses PmWiki but I don't think the rest of the arduino.cc uses it. I put the CapacitiveSensor code in a SourceBlock as documented here: http://www.pmwiki.org/wiki/Cookbook/SourceBlock.
>The playground code looks double-spaced

I also don't like the double spacing. This was done automatically by the SourceBlock and I don't see a way to change it. I notice now that whatever agdl did fixed the problem that was causing the code to be incorrectly formatted without a SourceBlock and when it is taken out of the SourceBlock the double spacing goes away so I can revert my edit if you like it better the way it was before.
>doesn't have the nice syntax highlighting like the Shftln21

I specified C as the language `(:source lang=c:)`(SourceBlock [doesn't support](http://qbnz.com/highlighter/examples.php) Arduino as a language). I also tried C++ with the same results. This should provide highlighting but it doesn't so I don't know if SourceBlock isn't installed or if something isn't configured correctly.

ElectricRCAircraftGuy, 2015-09-14:
@per1234 , Hi, thanks for the info.  Actually just spent the last couple hrs doing some research on it myself, and I just went around and edited a handful of Playground pages myself to improve the documentation on editing PmWiki pages.

See here: 
http://playground.arduino.cc/Main/Participate#code

I just added source code "Example 2". Notice how it differs from Example 1. Example 2 uses your source block lines including the [@....@] symbols. Without them, you get Example 1. 

I'm not sure which I like best exactly, but perhaps we should use the Example 1 style?

closed 2015-09-15 by ElectricRCAircraftGuy

per1234, 2015-09-15:
@ElectricRCAircraftGuy I don't like how Example 1 wraps text. Other than that I think Example 1 looks better than Example 2. So I prefer the Example 2 style unless there's a way to make Example 1 wrap but I'm not a contributor to the CapacitiveSensor code so I'm happy to go with your choice of 1, 2 or no SourceBlock.

ElectricRCAircraftGuy, 2015-09-15:
It's good enough as-is. If I change my mind I'll change it but I think it's fine. I'll prob. end up submitting a pull-request later on to add it to the Arduino core example too, and that will of course make it easiest to read.

...I"m still wondering how the web-page gets created for core examples though...perhaps someone creates it once the example is accepted into the core?