arduino/Arduino #3790: Example doesn't compile

dragospuri, 2015-09-13:
Using Arduino IDE 1.6.5
Installed the library through the manager. Opened the example. Doesn't compile.

Arduino: 1.6.5 (Windows 7), Board: "Arduino Nano, ATmega328"

Using library PrintEx in folder: C:\Users\Surreal\Documents\Arduino\libraries\PrintEx 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs -IC:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src C:\Users\Surreal\AppData\Local\Temp\build139473521994345729.tmp\basic_usage.cpp -o C:\Users\Surreal\AppData\Local\Temp\build139473521994345729.tmp\basic_usage.cpp.o 

In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:106:25: error: wrong number of template arguments (1, should be 2)
    !is_pointer<D>::value>::type
                         ^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:42:41: error: provided for 'template<bool V, class T> struct enable_if'
   template< bool V, typename T > struct enable_if{}; //C++98 must provide all template parameters.
                                         ^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:3: error: template argument 3 is invalid
   >{
   ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: expected '::' before '{' token
   >{
    ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: expected identifier before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:107:4: error: qualified name does not name a class before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:119:24: error: wrong number of template arguments (1, should be 2)
    is_pointer<D>::value>::type
                        ^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:42:41: error: provided for 'template<bool V, class T> struct enable_if'
   template< bool V, typename T > struct enable_if{}; //C++98 must provide all template parameters.
                                         ^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:8:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:3: error: template argument 3 is invalid
   >{ 
   ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: expected '::' before '{' token
   >{ 
    ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: expected identifier before '{' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/TypeTraits.h:120:4: error: qualified name does not name a class before '{' token
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintExtension.h:26:0,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/MemoryPrinter.h:9,
                 from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:8,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:19:34: warning: variadic templates only available with -std=c++11 or -std=gnu++11 [enabled by default]
    template< typename T, typename... U >
                                  ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/PrintVariadic.h:20:46: warning: variadic templates only available with -std=c++11 or -std=gnu++11 [enabled by default]
     derived &printx( const T& t, const U&... u ){
                                              ^
In file included from C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/PrintEx.h:9:0,
                 from basic_usage.ino:8:
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected template-name before '<' token
   ios::OStreamBase<StreamExtension>,
                   ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected '{' before '<' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:20:19: error: expected unqualified-id before '<' token
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:27:23: error: invalid use of incomplete type 'struct ios::StreamExtension'
     struct StreamEx : StreamExtension{
                       ^
C:\Users\Surreal\Documents\Arduino\libraries\PrintEx\src/lib/StreamExtension.h:16:9: error: forward declaration of 'struct ios::StreamExtension'
  struct StreamExtension : 
         ^
basic_usage.ino:48:14: error: expected '}' at end of input
expected '}' at end of input

q2dg, 2015-09-13:
It seems an error of PrintEx library. Please close this issue and open it in https://github.com/Chris--A/PrintEx

closed 2015-09-13 by dragospuri

Chris--A, 2015-09-13:
This is due to C++11 not being active, even though the commit history reflects it, I opened an issue here: https://github.com/arduino/Arduino/issues/3791 to clarify what is going on.

This issue has been moved here: https://github.com/Chris--A/PrintEx/issues/6