arduino/Arduino #3791: C++11 was added July 16, but not found in latest release.

Chris--A, 2015-09-13:
@cmaglie @matthijskooijman 

Hi, I'm a little confused about the history of commits.

On July 16 C++11/C11 was enabled. However the latest release does not reflect what is in the repo.

Is there something I'm missing here as I was expecting it to be available. 
https://github.com/arduino/Arduino/pull/2175 / ( https://github.com/arduino/Arduino/commit/7fd62446213cc395f9cc3f886886107cbfb6ba5e / https://github.com/arduino/Arduino/commit/ddf4d878fdb39173b872f69149fb4cf1cd9fd017 )

Has it been removed again until 1.6.6? (history does not reflect this).

matthijskooijman, 2015-09-13:
@Chris--A, The 1.6.5 release was made in june, before this PR was merged. 1.6.5-r3 was released after, but AFAIK the -rx releases only have minor, platform-specific changes to the packaging etc., they are not built from the master revision, but from the original released version (1.6.5 in this case) with some changes applied. For example, here's the history for the 1.6.5-r3 tag, that reflects this: https://github.com/arduino/Arduino/commits/1.6.5-r3

Does this clarify your question? If so, please close this issue.

Chris--A, 2015-09-13:
Yes that explains it, I assumed the latest revisions would include the commits up to date. I was not aware they only contain selected fixes.

closed 2015-09-13 by Chris--A

q2dg, 2015-09-14:
So...it's time for a 1.6.6, no?

Chris--A, 2015-09-15:
Or at least an unfinished beta so the community can test the huge amount of changes.

matthijskooijman, 2015-09-15:
@Chris--A, There is of course the hourly builds, which are built from git master. Were you aware of those? So far, I've pointed people to then when I needed C++11 or C99 support for some code.

Chris--A, 2015-09-15:
Yes, I keep my git fork up to date and build my own frequently, which is why I assumed the features are present.

However it has been shown time and time again that there is not enough of us using the latest build of the IDE in a vast variety of ways to discover all the potential pitfalls and bugs. Essentially we all lack the '*ability*' to see and use the IDE from a newbies perspective.  Proof of concept/beta testing should not be left to the developers, and involving people who have never used the IDE would allow for the best possible outcome (with over a million downloads, there are new potential testers every day).

matthijskooijman, 2015-09-15:
@Chris--A, Not sure if you understood what I meant. Instead of building from git yourself, you can download hourly builds from arduino.cc. Installing those is identical to normal releases (except they maybe lack signed drivers, dunno), so they should be usable as testing versions for a broad audience. Still, having an actual "beta" version released would probably attract more testers than the nightly builds currently, that is true.

Chris--A, 2015-09-15:
I understand what you're putting forward, however like you say, a beta which is spearheaded by the Arduino team with a dedicated thread on the forum for responses would be far more efficient. 

On many of the computers I have installed the IDE on, the hourly build section (downloads page) is below the fold and never going to be noticed by a majority of users.