arduino/Arduino #3793: Difficult to synchronise libraries across multiple code revisions

NanoAkron, 2015-09-15:
Hi all,

Been using Arduino now for years. Problem is I like to have my libraries open alongside my main program during development. And I keep different versions as I go along.

So...my directory structure ends up quite messy, with multiple copies of my libraries at various stages of development as my project progresses.

Is there any way you could let me just select a 'show libraries' button so I can keep my libraries in the main library folder yet still have them ready to view and edit?

Chris--A, 2015-09-15:
Use git for version control, then you never need to move them. Just switch branches. The compilation process grabs the whole directory for each library so its best to keep a single version accessible at a time. 

The main library folder being documents\Arduino\libraries? I'm not sure I follow your second issue. I develop all my libraries from the main library folder. I have my lib open in N++, my sketch in the IDE, and Git running in the background.  A desktop/taskbar shortcut is what you're looking for, no?

NanoAkron, 2015-09-15:
You assume I'm technical enough to install and use git, or that thousands of others are. Why can't the Arduino IDE just take care of this for me?

Load up my sketch, select 'show libraries' or something like that, and get them opened up at the same time.

Chris--A, 2015-09-15:
By looking at your profile, which I did before posting, game me the impression that you would at least have the basic usage of Git/GitHub IDE already sussed out. Regardless of your capabilities, the IDE is not designed to look after version control of custom libraries. It can only update/revert libraries installed via the library manager.

The IDE's role is to build sketches, libraries are essentially pre-built expansions. Editing these from the IDE has been discussed, and a possibility for the future, however it is a project based environment requiring an .ino file.

You can create your library as a sketch first then move it over to the library folder. Having the IDE show the libraries may be useful, but creating your own desktop shortcut is sufficient, and is something that is incredibly easy to find information on how to do. You also need to realize that there is more than one location for libraries.

If you want to know what is installed, the library manager will provide this information for you too.

NanoAkron, 2015-09-16:
If library management isn't meant to be a function of the IDE then I understand.

ffissore, 2015-09-25:
Related to #3512. In particular see https://github.com/arduino/Arduino/issues/3512#issuecomment-128029263. As @Chris--A said, as of now, the IDE is a tool for building sketches. With `.development` file you can at least develop your library examples without having the IDE complain because they are read-only. Still, library code development is not handled.

closed 2015-09-25 by ffissore

lmihalkovic, 2015-10-31:
Had the issue that my tabs were not visible because I had too many of them. Needed to be able to search inside the source of libraries as well a my code. Wanted to write my code cleanly as libraries... short version:  https://github.com/arduino/Arduino/issues/4050. 

I keep the libs code in a separate folder and symlink them into "sketchabc/libraries/LIBxxxx" "sketchdef/libraries/LIBxxxx". Then in the editor I can open sketchabc and see/edit the source code for the sketch as well as for LIBxxxx. 

seems like a workable workflow so far..

lmihalkovic, 2016-01-21:
@NanoAkron I learn a lot by reading source code and agree that IDE can provide assistance (not yet available, but will be)

[image]