arduino/Arduino #3794: Driver Issues with Windows 10

RileyRobinson4, 2015-09-15:
I'm running Windows 10 and i having trouble getting the drivers to work for my Arduino Nano. The nano is connected to the COM3 port, but the Arduino application only recognizes COM2 under the Tools tab. I have tried reinstalling the application, restarting my computer, and manually installing the drivers, but the nano is not being recognized. I downloaded the drivers from https://www.arduino.cc/en/Main/Software. Under the Hardware settings for the COM3 Port, it says. Device Status: Needs Troubleshooting, Driver Error, and This Device Cannot Start (code 10). I have tried everything that I could think of, and nothing works. Please help me.

NicoHood, 2015-09-15:
Win 10 doesnt need any drivers.

RileyRobinson4, 2015-09-15:
I believe the issue was that i have a sunfounder nano, and it takes a special driver. I haven't tried i out yet, but the Sunfounder technical support man said it was most likely my issue.

q2dg, 2015-09-15:
That's not an official board, then. I don't think you'll be able to get support from this place, sorry

facchinm, 2015-09-22:
Just tested on Win10 and it works great with Arduino/Genuino boards. Surely your problem is related with extra drivers needed due to unofficial board

closed 2015-09-22 by facchinm