arduino/Arduino #3795: Entry for newly added URL doesn't appear the first time Boards Manager is opened

per1234, 2015-09-15:
Using Arduino IDE 1.6.6 2015/09/14 05:43 with Windows 7 64 bit
- Add a URL for a package that has never been installed to File>Preferences>Additional Boards Manager URLs
- Tools>Board>Boards Manager...
- Wait for all downloads to complete
- The entry for the newly added URL does not appear in Boards Manager
- Close Boards Manager
- Tools>Board>Boards Manager...
- Wait for all downloads to complete
- The entry for the newly added URL appears in Boards Manager

This happens consistently. The issue does not occur with 1.6.5r5

NicoHood, 2015-09-17:
I remember to have the same problem.

SpenceKonde, 2015-11-04:
This is really confusing to new users. I've just had to amend the installation instructions for the cores I maintain to reflect this bug.

ffissore, 2015-11-17:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-17 by ffissore

hatam1975, 2020-02-19:
hhhhh