arduino/Arduino #3796: Hourly build seem to have removed  -mmcu={build.mcu} argument to compiler

fallberg, 2015-09-15:
This causes a lot of problems for low level functionality. For example, sleep.h gets a preprocessor error because of missing precompiler defines obtained from the -mmcu compiler argument.

ffissore, 2015-09-15:
Thanks for the heads up! Fixing it right now

ffissore, 2015-09-15:
uhm, can you provide the command line you're using? I've just tried compiling a Blink for a Nano with 
```
./arduino --verify --board arduino:avr:nano:cpu=atmega328 --verbose ./examples/01.Basics/Blink/Blink.ino 
```
and it works as expected

fallberg, 2015-09-15:
I use the GUI to test. Blink compiles, since it does not include AVR specific stuf (at least not sleep.h). Just try to add #include <avr\sleep.h> and it will fail (at least for an Arduino Nano which is the board I tested with).

ffissore, 2015-09-15:
Confirm. Thanks

ffissore, 2015-09-15:
Fixed in latest hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-09-15 by ffissore