arduino/Arduino #3797: [DOC] Vagueness in data types explanation

q2dg, 2015-09-15:
In https://www.arduino.cc/en/Reference/Word is written: "A word stores a 16-bit unsigned number, from 0 to 65535. Same as an unsigned int. " But in https://www.arduino.cc/en/Reference/UnsignedInt is written "The Due stores a 4 byte (32-bit) value, ranging from 0 to 4,294,967,295 (2^32 - 1)." So...

*Does the Due also store 4 bytes in the word type?
*If so...maybe it should be noted that then, in Due the word type and the unsigned long type are the same and also the int type and the long type.
*What about the Zero? It's the same than the Due?

In https://www.arduino.cc/en/Reference/Double only mentions Due has 8-byte precision. Zero too?

VogonJeltz, 2015-09-15:
I don't underatnd you question - AFAIK, on most 8bit cpus
16 bit values are short, word, int
32 bit values are long

on 32 bit system mostly
16 bit values are short and word
32 bit values are int or long
64 bit values are long long

moreover,
on Arduino AVR char is 8-bit signed, while
on the Due char is 8-bit UNSIGNED !

Because all these values are ambiguous I'm only useing the exact-size values
int8_t, uint8_t
int16_t, uint16_t
int32_t, uint32_t

for char in strings to be used for Serial it's not possible though in these cases for all platform only char works, even if its signed for AVRs and unsigned for Due.

This indeed is a bug.

q2dg, 2015-09-15:
Well, I'll rewrite the issue putting "?" where I consider there's no a clear statement. 
According to "Data Types" pages from https://www.arduino.cc/en/Reference/HomePage:

char :                           1 signed byte (AVR & ARM)      --> It's clear
byte (unsigned char) : 1 unsigned byte (AVR & ARM)  --> It's clear
int :                              2 signed bytes (AVR) ; 4 signed bytes (Due and...**Zero??**)
short :                          2 signed bytes (AVR & ARM             --> so for AVR it's the same than int)
word (unsigned int) :   2 unsigned bytes (AVR) ; 4 unsigned bytes (Due and...**Zero??**)
long :                           4 signed bytes (AVR & **ARM??**     --> so for ARM it'd be the same than int)
unsigned long :            4 unsigned bytes (AVR & **ARM??** --> so for ARM it'd be the same than word)
float :                            4 bytes (AVR & ARM)
double :                        4 bytes (AVR) ; 8 bytes (Due and...**Zero??**)

VogonJeltz, 2015-09-15:
char is unsigned (!) for ARM !!
word is always 2 bytes !

int is 16 bit on AVR and Due
long is 32 bit on AVR and ARM 
double is not double on AVR, its just a #define for float on Arduinos, 
double is always 2x precision of float - except for Arduino AVR workarounds.

q2dg, 2015-09-15:
Well, from https://www.arduino.cc/en/Reference/Char: "The char datatype is a signed type, meaning that it encodes numbers from -128 to 127. For an unsigned, one-byte (8 bit) data type, use the byte data type". It not distinct AVR from ARM. Who is right, then???

And as I said at first message: in https://www.arduino.cc/en/Reference/Word is written: "A word stores a 16-bit unsigned number, from 0 to 65535. Same as an unsigned int." But in https://www.arduino.cc/en/Reference/UnsignedInt is written "The Due stores a 4 byte (32-bit) value, ranging from 0 to 4,294,967,295 (2^32 - 1)." So I infer Due (and Zero?) also store 4 bytes in the word type. Who is right, then??

VogonJeltz, 2015-09-15:
try it: on Athe DUE char is  definitely unsigned!
I ran in many issues because of that!

q2dg, 2015-09-15:
Well, tha't why I've opened this issue...to clarify the use of the right words in defining the data types on both architectures

VogonJeltz, 2015-09-15:
ok, what IS is easy to find out:

~~~

//=====================================================================================
//=====================================================================================
void setup() {
   char sbuf[128];   
   int32_t  i=0;
         
   // Serial terminal window
   i=115200;
   Serial.begin(i);  

}

//=====================================================================================
void loop(){
   char     sbuf[128];

  sprintf(sbuf, "sizeof(char) == %d", sizeof(char)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(short) == %d", sizeof(short)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(word) == %d", sizeof(word)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(int) == %d", sizeof(int)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(long) == %d", sizeof(long)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(float) == %d", sizeof(float)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(double) == %d", sizeof(double)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(long double) == %d", sizeof(long double)); Serial.println(sbuf);
  sprintf(sbuf, "sizeof(long long) == %d", sizeof(long long)); Serial.println(sbuf);

   while(1);
  
   

}
//=====================================================================================
//=====================================================================================
~~~

on Mega:

sizeof(char) == 1
sizeof(short) == 2
sizeof(word) == 2
sizeof(int) == 2
sizeof(long) == 4
sizeof(float) == 4
sizeof(double) == 4
sizeof(long double) == 4
sizeof(long long) == 8

on Due:

sizeof(char) == 1
sizeof(short) == 2
sizeof(word) == 4
sizeof(int) == 4
sizeof(long) == 4
sizeof(float) == 4
sizeof(double) == 8
sizeof(long double) == 8
sizeof(long long) == 8

VogonJeltz, 2015-09-15:
but that DOES NOT APPLY TO C STANDARDS!

IMO Arduino has to fix that, esp. for word, long, double !

~~~
Here is the complete list of fundamental types in C++:
Group	Type names*	Notes on size / precision
Character types	char	Exactly one byte in size. At least 8 bits.
char16_t	Not smaller than char. At least 16 bits.
char32_t	Not smaller than char16_t. At least 32 bits.
wchar_t	Can represent the largest supported character set.
Integer types (signed)	signed char	Same size as char. At least 8 bits.
signed short int	Not smaller than char. At least 16 bits.
signed int	Not smaller than short. At least 16 bits.
signed long int	Not smaller than int. At least 32 bits.
signed long long int	Not smaller than long. At least 64 bits.
Integer types (unsigned)	unsigned char	(same size as their signed counterparts)
unsigned short int
unsigned int
unsigned long int
unsigned long long int
Floating-point types	float	
double	Precision not less than float
long double	Precision not less than double
Boolean type	bool	
Void type	void	no storage
Null pointer	decltype(nullptr)	
~~~

q2dg, 2015-09-16:
Well, if you don't mind, i'll close this issue and I'll reopen a new one with a more concise text as this one has become a little hard to follow

closed 2015-09-16 by q2dg