arduino/Arduino #3798: Version 16h57.22 of 16/9/2015 error issue

pierrotm777, 2015-09-15:
If i try to compile the 'Blink' example , the IDE return me an error:
E:\Téléchargements terminés\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware="E:\Téléchargements terminés\arduino-nightly\hardware" -hardware="C:\Users\pierre\AppData\Local\Arduino15\packages" -tools="E:\Téléchargements terminés\arduino-nightly\tools-builder" -tools="E:\Téléchargements terminés\arduino-nightly\hardware\tools\avr" -tools="C:\Users\pierre\AppData\Local\Arduino15\packages" -libraries="C:\tmp\Arduino_Compilateur\arduino-1.6.6\MyArduinoSketch\MesProgrammes\libraries" -libraries="E:\Téléchargements terminés\arduino-nightly\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path="C:\Users\pierre\AppData\Local\Temp\build6781017315140768274.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "E:\Téléchargements terminés\arduino-nightly\examples\01.Basics\Blink\Blink.ino"
Usage of E:\TÃ©lÃ©chargements terminÃ©s\arduino-nightly\arduino-builder:
  -build-path="": build path
  -compile=false: compiles the given sketch
  -debug-level=5: Turns on debugging messages. The higher, the chattier
  -dump-prefs=false: dumps build properties used when compiling
  -fqbn="": fully qualified board name
  -hardware=[]: Specify a 'hardware' folder. Can be added multiple times for specifying multiple 'hardware' folders
  -ide-version="10600": fake IDE version
  -libraries=[]: Specify a 'libraries' folder. Can be added multiple times for specifying multiple 'libraries' folders
  -logger="human": Sets type of logger. Available values are 'human', 'machine'
  -prefs=[]: Specify a custom preference. Can be added multiple times for specifying multiple custom preferences
  -tools=[]: Specify a 'tools' folder. Can be added multiple times for specifying multiple 'tools' folders
  -verbose=false: if 'true' prints lots of stuff
  -version=false: prints version and exits
  -warnings="": Sets warnings level. Available values are 'none', 'default', 'more' and 'all'

Erreur lors de la compilation.

pierrotm777, 2015-09-15:
With another code, the IDE return me:
C:\tmp\Arduino_Compilateur\arduino-1.6.6\arduino-builder -dump-prefs -logger=machine -hardware="C:\tmp\Arduino_Compilateur\arduino-1.6.6\hardware" -hardware="C:\Users\pierre\AppData\Local\Arduino15\packages" -tools="C:\tmp\Arduino_Compilateur\arduino-1.6.6\tools-builder" -tools="C:\tmp\Arduino_Compilateur\arduino-1.6.6\hardware\tools\avr" -tools="C:\Users\pierre\AppData\Local\Arduino15\packages" -libraries="C:\tmp\Arduino_Compilateur\arduino-1.6.6\MyArduinoSketch\MesProgrammes\libraries" -libraries="C:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path="C:\Users\pierre\AppData\Local\Temp\build5390521111426916597.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "E:\Téléchargements terminés\arduino-nightly\examples\01.Basics\Blink\Blink.ino"
panic: runtime error: index out of range

goroutine 1 [running]:
arduino.cc/builder/props.loadSingleLine(0x121548a0, 0x1216c000, 0x39)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/props/properties.go:92 +0x1a0
arduino.cc/builder/props.Load(0x1210f1d0, 0x4b, 0x2, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/props/properties.go:70 +0x1f2
arduino.cc/builder.loadBoards(0x12154840, 0x121025c0, 0xa, 0x12152696, 0x3, 0x12106880, 0x40, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/hardware_loader.go:222 +0xc0
arduino.cc/builder.loadPlatform(0x12154820, 0x121025c0, 0xa, 0x12106880, 0x40, 0x12154760, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/hardware_loader.go:183 +0x1d2
arduino.cc/builder.loadPackage(0x12152660, 0x12106500, 0x3c, 0x12109da0, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/hardware_loader.go:147 +0x68a
arduino.cc/builder.(*HardwareLoader).Run(0x617b40, 0x121083a0, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/hardware_loader.go:90 +0xa91
arduino.cc/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x617b40, 0x121083a0, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/container_setup.go:60 +0x500
arduino.cc/builder.runCommands(0x121083a0, 0x1214ded0, 0x2, 0x2, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/builder.go:133 +0xea
arduino.cc/builder.(*ParseHardwareAndDumpBuildProperties).Run(0x1214def4, 0x121083a0, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/builder.go:122 +0x105
arduino.cc/builder.RunParseHardwareAndDumpBuildProperties(0x121083a0, 0x0, 0x0)
	c:/jenkins/workspace/arduino-builder-windows/src/arduino.cc/builder/builder.go:165 +0x44
main.main()
	c:/jenkins/workspace/arduino-builder-windows/main.go:254 +0x16b2
arduino-builder a retourné 2

Erreur lors de la compilation.

ffissore, 2015-09-16:
This issue was moved to arduino/arduino-builder#7

ffissore, 2015-09-16:
Turns out this is an integration issue. We can solve it by changing **how** we call arduino-builder

ffissore, 2015-09-16:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 6847c524 (2015-09-16): arduino-builder arguments must be passed in separate cmd.addArgument calls, otherwise special chars will break it. Fixes #3798
closed 2015-09-16 by ffissore