arduino/Arduino #3804: New preprocessor/builder does not provide per-processor compiler definitions

Wackerbarth, 2015-09-16:
Based on the -mmcu= option to the compiler, the gcc compiler defines some preprocessor items which can be utilized to provide alternate compilations (For example, a token like __AVR_ATmega2560__ is defined for each processor. There are also tokens like USBCON which are defined for all of the mcus which have built-in USB interfaces.

There is a significant body of code which uses these tokens to customize a library.

AFAICT, the preprocessor in the new arduino-builder no longer has access to these items.

matthijskooijman, 2015-09-16:
This seems to be a duplicate of #3976, which was fixed yesterday. Could you please try the latest hourly build to see if it is solved for you too? I'm closing this issue, but please leave a comment with your findings, so we can re-open if needed.

closed 2015-09-16 by matthijskooijman

Wackerbarth, 2015-09-17:
Recent changes seem to have resolved this issue.