arduino/Arduino #3805: StandardFirmata don't compile

pierrotm777, 2015-09-16:
The example StandardFirmata return me an issue when i compile !
My board is an Uno.

"C:\tmp\Arduino_Compilateur\arduino-1.6.6\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "C:\Users\pierre\AppData\Local\Temp\build9193688633695281333.tmp\sketch\StandardFirmata.ino.cpp"
Utilisation de la bibliothèque Servo version 1.1.0 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Servo 
Utilisation de la bibliothèque Wire version 1.0 dans le dossier: C:\Users\pierre\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\libraries\Wire 
Utilisation de la bibliothèque Firmata version 2.4.4 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Firmata 
"C:\tmp\Arduino_Compilateur\arduino-1.6.6\hardware\tools\avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR         "-IC:\Users\pierre\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\cores\arduino" "-IC:\Users\pierre\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\variants\standard" "-IC:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Servo\src" "-IC:\Users\pierre\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\libraries\Wire" "-IC:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Firmata" "C:\Users\pierre\AppData\Local\Temp\build9193688633695281333.tmp\sketch\StandardFirmata.ino.cpp"
In file included from C:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Firmata/Firmata.h:16:0,

                 from StandardFirmata.ino:28:

C:\tmp\Arduino_Compilateur\arduino-1.6.6\libraries\Firmata/Boards.h:529:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"

 #error "Please edit Boards.h with a hardware abstraction for this board"

  ^

exit status 1
Erreur lors de la compilation.

This is an issue  with last hourly ide 16 September 2015 16:36:14 GMT

NicoHood, 2015-09-17:
And the library also throws a warning in my IDE with an incompatible library manager. Very annoying.

ffissore, 2015-09-17:
I see you're using 1.6.6 but you're also using avr core installed through boards manager. Delete folder `C:\Users\pierre\AppData\Local\Arduino15\packages\arduino\hardware\avr` and retry.

pierrotm777, 2015-09-17:
Ho, i see two different arduino15 folder:
C:\Users\pierre\AppData\Roaming\arduino15 , the last
C:\Users\pierre\AppData\Local\Arduino15 , i think an old !

???

ffissore, 2015-09-17:
(y)

closed 2015-09-17 by ffissore