arduino/Arduino #3806: LCD hello world

noisegate, 2015-09-16:
In this tutorial https://www.arduino.cc/en/Tutorial/HelloWorld, it says in the code comment to connect LCD module pin r/w (5) to GND. In the (Fritzing) schematics it is connected to +5V. I think the fromer is correct (pin r/w (5) on the LCD module should be connected to GND). 
could you look into this please, thx in advance
code:
```c
/*
  LiquidCrystal Library - Hello World

 Demonstrates the use a 16x2 LCD display.  The LiquidCrystal
 library works with all LCD displays that are compatible with the
 Hitachi HD44780 driver. There are many of them out there, and you
 can usually tell them by the 16-pin interface.

 This sketch prints "Hello World!" to the LCD
 and shows the time.

  The circuit:
 * LCD RS pin to digital pin 12
 * LCD Enable pin to digital pin 11
 * LCD D4 pin to digital pin 5
 * LCD D5 pin to digital pin 4
 * LCD D6 pin to digital pin 3
 * LCD D7 pin to digital pin 2
 * LCD R/W pin to ground
 * LCD VSS pin to ground
 * LCD VCC pin to 5V
 * 10K resistor:
 * ends to +5V and ground
 * wiper to LCD VO pin (pin 3)

 Library originally added 18 Apr 2008
 by David A. Mellis
 library modified 5 Jul 2009
 by Limor Fried (http://www.ladyada.net)
 example added 9 Jul 2009
 by Tom Igoe
 modified 22 Nov 2010
 by Tom Igoe

 This example code is in the public domain.

 http://www.arduino.cc/en/Tutorial/LiquidCrystal
 */

```

[image]

agdl, 2015-09-17:
Duplicate of#3810

closed 2015-09-17 by agdl

noisegate, 2015-09-17:
yeah 3806 > 3810? Thanks for looking into it though..