arduino/Arduino #3807: arduino target file is in hardcoded folder sketch

jantje, 2015-09-16:
as @matthijskooijman states:

>would it perhaps be an option to store the source and .o files inside sketch, but put the final linked .elf and .hex files directly in the build folder? This might fix most backward compatibility problems I believe, and it would still make sense (and not cause problems)?

the name "sketch" is hardcoded. That means that everybody making tools on top of the platform.txt needs to parse the content of the value of recipe.c.combine.pattern recipe.objcopy.eep.pattern and many more yust to know where the target file is located or hardcode multiple locations as proposed by @ffissore .
I think this is a very bad approach. At least the name sketch should be in a variable but even better the target should be in the {build.path} folder.

PS we have had a very similar discussion about the archive name which was also hardcoded in the one of the recipes

matthijskooijman, 2015-09-16:
Originally, all of the sketch (so source and object files, as well as .elf/.hex/.bin/eep products) were in build.path. To prevent conflicts with libraries and the core, the sketch files were moved into sketch. However, we could consider putting the source and object files (which can have arbitrary names) into the sketch folder, and keep the build products direcly in build.path. If you do this, then I believe that the sketch subfolder sort of becomes an implementation detail of platform.txt?

 IIRC they are now named e.g. name_of_sketch.elf, it might be even easier if they'd also be called sketch.elf (so literally using "sketch"), so it is even easier to find the build products?

Doing the (first part) of the above would fix backward compatibility with earlier versions, though it might actually break compatibility with the most recent version again. If we want to do this, it is something to do *now*, before another version is released that has the changed build product location.

matthijskooijman, 2015-09-17:
It actually seems that the SAMD core already uses the proposed approach:
```
/home/matthijs/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc -L/tmp/build6421611539749575012.tmp -Os -Wl,--gc-sections -save-temps -T/home/matthijs/.arduino15/packages/arduino/hardware/samd/1.6.1/variants/arduino_zero/linker_scripts/gcc/flash_with_bootloader.ld -Wl,-Map,/tmp/build6421611539749575012.tmp/Blink.cpp.map --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m0plus -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o /tmp/build6421611539749575012.tmp/Blink.cpp.elf /tmp/build6421611539749575012.tmp/sketch/Blink.cpp.o /tmp/build6421611539749575012.tmp/core/variant.cpp.o -Wl,--start-group -lm /tmp/build6421611539749575012.tmp/core/core.a -Wl,--end-group 
/home/matthijs/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy -O binary /tmp/build6421611539749575012.tmp/Blink.cpp.elf /tmp/build6421611539749575012.tmp/Blink.cpp.bin 
```

In this output, not that the .o file is in the subfolder: `/tmp/build6421611539749575012.tmp/sketch/Blink.cpp.o` but the elf file is not: `/tmp/build6421611539749575012.tmp/Blink.cpp.elf `.

ffissore, 2015-09-17:
Assigned to @cmaglie /cc @facchinm and @sandeepmistry as, although I have changed the location of the final .hex, that is driven by platform.txt

cmaglie, 2015-09-17:
For me it's ok to put the final binary in the root build folder.

jantje, 2015-09-17:
:-)

cmaglie, 2015-09-18:
Here is the fix: https://github.com/arduino/Arduino/commit/e2f625290f996a01018de26af548d85a692354a0

closed 2015-09-18 by cmaglie

jantje, 2015-09-18:
Thank you very much.

commit e2f62529 (2015-09-18): Moved build artifacts into root build folder

The intermediate files used for build (.o, .a ...) are still
in subfolders to avoid conflicts.

Fix #3807