arduino/Arduino #3808: Crash: "no splash screen available"

matthijskooijman, 2015-09-16:
Just tried to start my Arduino, passing a single .ino file on the commandline, and got:
```
java.lang.IllegalStateException: no splash screen available
        at java.awt.SplashScreen.checkVisible(SplashScreen.java:197)
        at java.awt.SplashScreen.update(SplashScreen.java:324)
        at cc.arduino.view.SplashScreenHelper.splashText(SplashScreenHelper.java:82)
        at processing.app.Base.<init>(Base.java:450)
        at processing.app.Base.guardedMain(Base.java:223)
        at processing.app.Base.main(Base.java:139)
```

Haven't really investigated, but I didn't want to throw away the stacktrace. I'm running a self-compiled version from git master on Linux, so I can't rule out that there is something peculiar in my setup that causes this. The issue isn't reproducible, though, it worked (with a different .ino file) just before and it works (with the same .ino file) now, so it seems to be something random.

ffissore, 2015-09-17:
It's probably a timing issue: sometimes, by the time the IDE wants to print "starting" the splashscreen may be already gone. 57ed2b241696c1c3de02e499f6ab5f4d5cf604f9 should mitigate your issue, by checking visibility and then updating inside a synchronized block

closed 2015-09-17 by ffissore

matthijskooijman, 2015-09-17:
Awesome, thanks for the quick fix :-)

commit 57ed2b24 (2015-09-17): Updating splashscreen text ONLY if visible. Fixes #3808