arduino/Arduino #3809: IDE should not steal focus for "Update available for some of your libraries"

matthijskooijman, 2015-09-16:
Something that seems to be annoying: When the "Update available for some of your libraries" popup shows, the IDE window is focused (though not always, it seems). I'm not sure how this was designed to be, but ideally this popup would just stay within the IDE and should certainly not bring the entire IDE to the top.

Only tried on Linux, not sure how other platforms behave.

ffissore, 2015-09-17:
Sorry, I've tried but I was unable to avoid it from gaining focus. All OSs are affected. Help wanted

NicoHood, 2015-09-17:
Is there a way to disable auto update and manually search for new updates? That'd be at least an idea. Especially for dev versions when you get a new avr core, that you dont want to use.

i-make-robots, 2015-11-04:
Also the about dialog on startup refuses to stay down.
- start arduino
- see about image
- select any other window
- about image resurfaces
I can replicate this multiple times before app finishes loading.

sandeepmistry, 2016-07-18:
See https://github.com/arduino/Arduino/issues/4546#issuecomment-205312163, this should be resolved now.

closed 2016-07-18 by sandeepmistry