arduino/Arduino #3814: javaexception with todays (09/17/2015) version

harryboo, 2015-09-17:
Hi,

i have made some customizations to the menu like this (boards.txt):

menu.customvariant=Variant
.....
mighty.menu.customvariant.variantmighty=Mighty
mighty.menu.customvariant.variantmighty.build.variant=mighty
mighty.menu.customvariant.variantmighty.build.board=MIGHTY
mighty.menu.customvariant.variantavrdev=AVR Developer
mighty.menu.customvariant.variantavrdev.build.variant=mighty_avrdevelopers
mighty.menu.customvariant.variantavrdev.build.board=MIGHTY_AVRDEVELOPERS

All versions before there was no error.
After todays (09/17/2015) updates i got this exception:

Exception in thread "AWT-EventQueue-0" java.util.ConcurrentModificationException
	at java.util.LinkedList$LLSpliterator.forEachRemaining(LinkedList.java:1239)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
	at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:121)
	at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:105)
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:815)
	at processing.app.Base.onBoardOrPortChange(Base.java:1222)
	at processing.app.Base$14.actionPerformed(Base.java:1428)
	at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1467)
	at processing.app.Base.access$000(Base.java:79)
	at processing.app.Base$13.actionPerformed(Base.java:1400)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:308)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6535)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

matthijskooijman, 2015-09-18:
What did you do just before this exception occured? Does it occur everytime you do that, or only once?

harryboo, 2015-09-18:
I have updated the ide to the latest version and after loading a sketch i wanted to change something in the custom menu.
After massive testing the error occurs sometimes. I cannot say how to reproduce this.

ffissore, 2015-09-18:
I've pushed a small improvement (returning a copy of original list instead of the original one). While this is probably NOT the root cause of your issue, it may help. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly
Closing for now. Should the problem occur again, just reopen this issue. Thank you

closed 2015-09-18 by ffissore