arduino/Arduino #3816: OSX Arduino IDE 1.6.5 no indication when file is not saved

iseldonz, 2015-09-18:
Altering a file should put a black "bullet" in the red circle in the window's header. This does not happen, so ther is no way to tell whether an altered file has been saved.

ffissore, 2015-09-18:
This is when the sketch is saved
[image]
This is when is NOT saved
[image]
Where have you red about this black bulled in a red circle?

Chris--A, 2015-09-24:
I think @iseldonz is mentioning this:

[image]

Whether or not this is a common behavior, I do not know (do not use OSX)

Chris--A, 2015-09-24:
It is also mentioned in the apple developer guides.
https://developer.apple.com/library/mac/documentation/Java/Conceptual/Java14Development/07-NativePlatformIntegration/NativePlatformIntegration.html (search for *Window-Modified Indicator*).
So I guess it is a standard way of doing things.

ffissore, 2015-09-24:
Yes I was just reading it. Thank you, seems feasible

ffissore, 2015-09-24:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit befd7d42 (2015-09-24): OSX: attempting to draw a black dot in the upper left red button. Fixes #3816
closed 2015-09-24 by ffissore