arduino/Arduino #3822: Serial buffer size should depend on RAMEND - RAMSTART rather than RAMEND

NicoHood, 2015-09-19:
```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(RAMEND);
  Serial.println(RAMEND, HEX);
  Serial.println(RAMEND - RAMSTART);
  Serial.println(RAMEND - RAMSTART, HEX);
  delay(1000);

  // 32u4
  //2815
  //AFF
  //2559
  //9FF

  // 32u2
  //1279
  //4FF
  //1023
  //3FF

  //328
  //2303
  //8FF
  //2047
  //7FF
}
```

**Affected files:**
* https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.h#L42-L55
* https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBAPI.h#L78-L82
Maybe other files are also affected.

**Fix:**

```cpp
#if ((RAMEND - RAMSTART) < 1023)
#define SERIAL_BUFFER_SIZE 16
#else
#define SERIAL_BUFFER_SIZE 64
#endif
```

NicoHood, 2015-09-20:
I also suggest a methode that tells you the current RX or TX buffer size. I am working with the u2 series very often which only has 500 bytes of ram. An automated way to get the data via function and not via hardcoded definition name would be nice.

```cpp
uint16_t getBuffersizeRX(){
return SERIAL_BUFFER_SIZE;
}
```

I currently also need this for my RawHID. A general name for all libraries would be appreciated.

matthijskooijman, 2015-09-21:
Is there any particular reason you need the `-RAMSTART` stuff? It seems that RAMSTART is always 0x100? And even it isn't, the exact number of RAM bytes isn't really important here AFAICS, it's just used to distinguish between a small chip and a large chip, which works just fine by just looking at RAMEND. Using both is more correct, perhaps, so I don't mind the change.

Regarding the `getBuffersizeRx()`, that is a completely different issue. You should probably open up a new issue about this (though I remember this has been discussed before, so look for existing issues and discussions on the mailing list first).

facchinm, 2015-09-21:
Closed by #3826

closed 2015-09-21 by facchinm

NicoHood, 2015-09-22:
RAMSTART is not always 0x100 and may be 0x200 as well. You never know, and this should avoid any errors and shows other users how to do it correct. Thanks for merging.

`getBuffersizeRx()` Was just a quick idea, discuss it at the mailing list if you like to, I am not there as you might know. I dont want to open so many issues is such a short time, you know ;)

DD-DINESH, 2021-04-05:
Hi 
I am new to this concept 
i have a simple website(html form) which running on nodemcu. 
I need to perform Buffer overflow in nodemcu esp8266 12e, make the website unavilable.
after that i need to clear the buffer overflow problem, and make the website available...

it's possible to do that ? if Yes,Please tell me how can i do that? 

i am using audrino ide for this project. 

Notes: using setrxbuffersize() if i specified the buffer size and how do i know buffer is full and again set the buffer to normal size or increase the size 

please help me ....... 

i am sorry if i wrong anywhere!