arduino/Arduino #3824: Problems with version 1.6.4 under Mac Os

jackbell16, 2015-09-20:
Hi,
I'm trying to run the Arduino IDE 1.6.4 on OS X Yosemite10.10.5 with the latest version of Java, but once I opened the application, I got the loading phase of it, but after that it closes without any label or warning: what could be the problem ?

ffissore, 2015-09-21:
Can you try the latest 1.6.5? http://arduino.cc/en/Main/Software
If it still fails, please try starting the ide from the terminal
`PATH_TO_IDE/Contents/MacOS/Arduino`
It will print the cause of the problem. Please paste the whole error message here

jackbell16, 2015-09-21:
/Applications/Arduino.app/Contents/MacOS/Arduino
java.lang.NoSuchMethodError: processing.app.Base.getSettingsFile(Ljava/lang/String;)Ljava/io/File;
	at template.tool.GistUploader.init(Unknown Source)
	at processing.app.Editor.addTools(Editor.java:929)
	at processing.app.Editor.buildToolsMenu(Editor.java:784)
	at processing.app.Editor.buildMenuBar(Editor.java:543)
	at processing.app.Editor.<init>(Editor.java:233)
	at processing.app.Base.handleOpen(Base.java:899)
	at processing.app.Base.handleOpen(Base.java:884)
	at processing.app.Base.handleOpen(Base.java:880)
	at processing.app.Base.handleNew(Base.java:768)
	at processing.app.Base.<init>(Base.java:477)
	at processing.app.Base.guardedMain(Base.java:226)
	at processing.app.Base.main(Base.java:137)

ffissore, 2015-09-21:
Problem is with GistUploader tool, probably located in SKETCHBOOK/tools. Remove it and restart the IDE. Please don't skip upgrading to IDE 1.6.5: it contains fixes to more than 400 issues

closed 2015-09-21 by ffissore

jackbell16, 2015-09-21:
Where is the SKETCHbook/tools library ? I don't know where to search it. Now I've switch to IDE 1.6.5

ffissore, 2015-09-21:
On your mac it should be Documents/Arduino/tools

jackbell16, 2015-09-21:
I've remove it, but now I got this problem:

Arduino:1.6.5 (Mac OS X), Scheda:"Arduino/Genuino Uno"

Cannot run program "{runtime.tools.avr-gcc.path}/bin/avr-g++" (in directory "."): error=2, No such file or directory