arduino/Arduino #3827: [Feature request] Highlight the word "TODO" in text editor

NicoHood, 2015-09-20:
Via the keywords.txt you can highlight different words.
What i wish is that you highlight the word "TODO" (and possibly some synonyms) red by default. A new preset is required for those colors (or maybe its there and noone knows). This way you could also highlight DEBUG and others.

Example in scratch editor of elementary OS:
[image]

Chris--A, 2015-09-20:
Highlighting in comments has just been removed. You can use the find tool to 'find' your TODO's

https://github.com/arduino/Arduino/issues/3532 / https://github.com/arduino/Arduino/commit/94d1829b871dff9db704455796d4a50b735bad27

NicoHood, 2015-09-20:
Hm you are right, that was mine :D 
Its just an idea, close it if you dont like it.

Chris--A, 2015-09-20:
I personally would like different colour comments.

In notepad++ you can get three different colourings.

`// blah`

`/* blah2 */`

`/** blah3 **/`

I'll have a look later, and see if this is possible in RSyntaxTA.

ffissore, 2015-09-21:
Yep, comments get no "Arduino" colouring since 94d1829 because of #3532. If you can out with a solution, it's more than welcome

lmihalkovic, 2015-11-21:
As an Eclipse user, I drive the most value out of TODOs by reviewing them as a list where I can click on any to jump to the code in question. 

Before diving into highlighting I figured it would be faster to add the following: typing "tags" or "tasks" will prompt a search through the sketch code (local libraries and assembly files included). The results are grouped by file. Possible improvement: cache the results and flush when memory is putting pressure or when the source file is altered (pending a lifecycle management of the editor contents)

[image]