arduino/Arduino #3828: real HID keyboard interface for stdin on the DUE host port?

VogonJeltz, 2015-09-20:
when will we have a real HID keyboard interface for stdin (stdio.h keyboard commands: getchar, gets, scanf, puts, ...) on the DUE host port?

peabo7, 2015-09-20:
Never.

Because all you demand is that someone do it for you, but you won't even lift a finger to try fixing it yourself.

Explaining that you aren't good enough to try is BS.  You aren't actually that stupid, try to figure it out and when you run into a problem, ask for help.

cmaglie, 2015-09-20:
Duplicate of #3630.

@VogonJeltz, I just want to make you notice that all your posts are sent to over 640 persons, asking repeatedly the same thing does **not** increase your chance to get help. The only thing that increase is the number of people that gets pissed off.

closed 2015-09-20 by cmaglie

VogonJeltz, 2015-09-20:
there was just no reply yet that it will be included in the 1.6.6 or 1.6.7 libs or what ever, that's why I ask.