arduino/Arduino #3830: Very incorrect timeout handling in pulseInLong

igendel, 2015-09-20:
The three While loops in pulseInLong (at wiring_pulse.c) increase "numloops" from 0 to "maxloops" as a way to check and enforce the timeout parameter. However, "maxloops" is defined in clock cycles, but each loop run takes much more than a single clock cycle.

This results in timeouts that are actually over 20 times longer than the user requests. 

Since this function uses micros() anyway for timing the pulse, I suggest replacing the numloops/maxloops method with checks against micros(). For instance:

// After function header and pin definitions...

	unsigned long maxMicros = micros() + timeout;

	// wait for any previous pulse to end
        while ((*portInputRegister(port) & bit) == stateMask)
            if (micros() >= maxMicros)
	            return 0;

// etc.

facchinm, 2015-09-21:
Woa! I totally missed it :smile:  Fixed by c7b27431a92
Thank you for pointing out this bug!

matthijskooijman, 2015-09-21:
The fix does not handle `micros()` overflow properly. If `maxMicros = micros() + timeout` overflows, then `micros() > maxMicros` will return true immediately.

Instead, try: `startMicros = micros();` and `if (micros() - startMicros > timeout)`, since then the overflow will happen correctly.

igendel, 2015-09-21:
True, matthijskooijman. I suggested my solution off the top of my head... the possibility of overflow should be taken into account as well. Thanks!

facchinm, 2015-09-21:
@matthijskooijman , you are totally correct. 
While applying your patch (https://github.com/facchinm/Arduino/commit/480cd227eab0c36e9c1eddc5c68c09589e500ab1), however, I encountered another issue.
Flash occupation increases by 72 bytes on AVR... Any cleaver idea on why this happens before starting a deeper analysis?

matthijskooijman, 2015-09-21:
Nothing springs to mind just now, but it does sound like something to investigate. If you want to investigate, have a look at this script I created to diff .elf file symbol tables (there is also a script to diff ASM output and reducing the amount of junk from differing addresses): https://github.com/matthijskooijman/scripts/blob/master/Arduino/diff-avr-sym

facchinm, 2015-09-25:
After a loooooong (way longer than a pulse :smile: ) analysis it turned out that the three 32bit arguments involved in the subtraction lead the compiler to allocate these variables in the stack instead than using registers, thus the +72 byte. 

Strangely, optimizing the function with `-O1` the size increases by only 24 bytes.

No math guru over the web came out with a better implementation so I'm merging @matthijskooijman's one with a note about this strange behaviour

closed 2015-09-25 by facchinm

matthijskooijman, 2015-09-26:
Ah, that makes sense - with my suggestion, the program needs to keep both  track of `startMicros` *and* `timeout`, while the previous code didn't need `timeout` any more after the initial calculation of `maxMicros`, which needs more variables. Not sure why O1 doesn't have this problem, perhaps it does less reordering of the code or something. Oh well.

PR #3864 (2015-09-25): fix pulseInLong considering overflow
fixes #3830

adding `__attribute__((optimize("O1")))` the total overhead is only 24 bytes (against +72 bytes with this implementation) (gcc 4.8.1)

For safety reasons this PR does not include this optimization
commit 480cd227 (2015-09-25): fix pulseInLong considering overflow

fixes #3830