arduino/Arduino #3831: Incorrect "Bootloader file specified but missing" warning

per1234, 2015-09-21:
Using Arduino IDE 1.6.6 Hourly Build 2015/09/18 06:03 with Windows 7 64bit
- Select any board with a bootloader specified that is not located under Arduino15/packages/arduino/hardware/avr
- Compile a sketch
- Output window displays the warning. For example: `Bootloader file specified but missing: C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\bootloaders\caterina\Caterina-Yun-noblink.hex`

ffissore, 2015-09-21:
Can you give the actual board you selected? If I select a Yun, no warning is printed

per1234, 2015-09-21:
It's Yun but I get it for any 3rd party boards also, for example Adafruit Metro. I think I know why we're getting different results. The Arduino AVR Boards package I'm using is located at \Arduino15\packages\arduino\hardware\avr\1.6.8 because I had updated the package from 1.6.7 previously. However when I deleted my Arduino15 and sketchbook/hardware folders to make sure no 3rd party hardware files were causing the issue I noticed that the IDE uses C:\Program Files (x86)\arduino-nightly\hardware\arduino\avr for Arduino AVR Boards v1.6.8 instead of using the Arduino15 folder. When I compile using that location I don't get the warning. So I installed Arduino AVR Boards v1.6.7 and the warning shows up again. So it happens only when the Arduino AVR Boards package is located in the Arduino15 folder and not when it's in the IDE install folder.

ffissore, 2015-09-21:
Ok then. This is the same issue as #3821. Bundled AVR core, while still published as 1.6.8, is really 1.6.9. In order to avoid a whole set of different configurations specific to the hourly builds, we don't use a greater version number for bundled AVR core.
So, when you're using bundled AVR core 1.6.8, you're actually using 1.6.9, while the core installed through boards manager is the "old" 1.6.8.
Assigning to @cmaglie as this will autofix once he'll release AVR 1.6.9

per1234, 2015-09-21:
Oh I see, sorry the Yun was a bad example, that bootloader actually is missing from the non-bundled 1.6.8 so the warning is correct in that case. I'm having the issue with some 3rd party hardware like [Mighty 1284P](https://github.com/JChristensen/mighty-1284p/tree/v1.6.3) but not with others like [Arduino on a Breadboard](https://www.arduino.cc/en/uploads/Tutorial/breadboard-1-6-x.zip). I haven't figured out what the difference is but I'll update if I find out.

per1234, 2015-09-26:
I figured out why Arduino on a Breadboard is not causing the warning to appear. A bootloader of the same name as that core specifies:
`atmega328bb.bootloader.file=atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex)`
is present in C:\Program Files (x86)\arduino-nightly\hardware\arduino\avr\bootloaders\atmega.  The issue is that the IDE is looking in the wrong location for the bootloader file when it does the check that causes the `Bootloader file specified but missing` message. I really don't think this is related to using the bundled AVR core because I still get this error if I delete the C:\Program Files (x86)\arduino-nightly\hardware\arduino\avr folder which should force the use of C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8 installed via Boards Manager which I'm guessing is the true Arduino AVR Boards 1.6.8 core.

To replicate this issue using Adafruit Flora as an example(must use Arduino IDE 1.6.6):
- Install Adafruit AVR Boards
  - Manual installation: https://github.com/adafruit/Adafruit_Arduino_Boards or:
  - Boards Manager URL: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
- Tools>Board>Adafruit Flora
- Compile any sketch - `Bootloader file specified but missing: C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\bootloaders\caterina\Caterina-Flora8.hex` warning appears in the output window even though the bootloader file is right where it should be.

per1234, 2015-09-28:
Fi\xed via https://github.com/arduino/arduino-builder/commit/d661e83adf7a956944e9f933484dd33bb5fd7226 thanks ffissore!

closed 2015-09-28 by per1234