arduino/Arduino #3832: Arduino does not support sketchbook on iCloud Drive?

M6KWH, 2015-09-21:
I would like to keep all my Arduino stuff - libraries, sketches etc, on iiCloud Drive folder.

But when browsing to set the sketchbook folder, iCould is not offered as a destination.

Or am I missing something?

ffissore, 2015-09-21:
Is your iCloud a real folder?

M6KWH, 2015-09-21:
Well it is the official Apple iCloud Drive folder visible in the Finder

ffissore, 2015-09-21:
I guess it's not a real folder then. I don't use any of these cloud stuff, but my colleague has his sketchbook synched with Dropbox

ffissore, 2015-09-21:
This stackoverflow may help you http://superuser.com/questions/827827/access-icloud-drive-via-terminal
Closing as invalid as this issue has nothing to do with the IDE, but it's a mac issue

closed 2015-09-21 by ffissore