arduino/Arduino #3836: ide crashes without notice if unprivileged user has cyrillic username on windows 8

mugiseyebrows, 2015-09-22:
I have `Пользователь` user account on windows 8 and arduino installer created directory `C:\Users\Џ®«м§®ў вҐ«м` with (`AppData` and `Documents` directories inside) which is obviously character encoding issue. When ide starts without elevated permissions it crashes without notice on "preparing boards" stage. It works if "run as administrator" used though. I run arduino_debug.exe and it shows "access denied" (отказано в доступе) error:
[image]
And it gives me a clue: `Пользователь` user account has no write permissions for `Џ®«м§®ў вҐ«м`. `icacls "C:\Users\?R<м§Rў в?<м" /grant Пользователь:(OI)(CI)F` (with administrator rights of course) solves this issue (I didnt type in `?R<м§Rў в?<м`, i tabbed it, dont know why its different this time). 
I encounter this issue on two machines with win8 so I guess its not just me.
Windows 8.1 x64, arduino 1.6.5 installed with installer, (I have jre 1.8.0_60 x64, jdk 1.8.0_60 x64 on this machine if that maters (I assume arduino uses its own java env))

ffissore, 2015-09-22:
thanks @mugiseyebrows, i'm looking into it

ffissore, 2015-09-23:
@mugiseyebrows I'm finding a hard time at trying to reproduce the issue. Is your windows in Russian?

mugiseyebrows, 2015-09-24:
Oh, sorry, I forgot to mention that. Yes, its in Russian.

mugiseyebrows, 2015-09-24:
I guess it's actually nsis issue. Is installation script in repository? I can't find it.

ffissore, 2015-09-25:
No, it does have to do with the installer. Apparently, Windows Registry is storing your username using a charset the IDE can't convert to UTF8, which is strange, since the native API is using wchar_t which is the correct type in this case (AFAIK). This one will take a while to solve and help from some windows expert is needed. In the meanwhile, create a `portable` folder in the IDE folder (something like `c:\program files\Arduino\portable`). The IDE will store everything but the sketchbook under `portable`: this will work around your problem **BUT** it will also mean that, next time you install the IDE, you will lose your settings, unless you back them up

mugiseyebrows, 2015-09-25:
Seems like duplicate of https://github.com/arduino/Arduino/issues/3360
CP866 -> CP1251 'Пользователь' -> 'Џ®«м§®ў вҐ«м'
Batch files on russian windows talks in CP866 by default, and other things talks in CP1251, which leads to confusions all the time.

ffissore, 2015-09-25:
#3360 was fixed by fetching data directly from the windows registry, instead of executing `regedit` from the command line.