arduino/Arduino #3841: Error in the description of the library EEPROM.get ()

pakhontas, 2015-09-22:
https://www.arduino.cc/en/Reference/EEPROMGet
Now:
  eeAddress = sizeof(float); //Move address to the next byte after float 'f'.
Need:
  eeAddress += sizeof(float); //Move address to the next byte after float 'f'.

Chris--A, 2015-09-22:
As the initial value is zero, adding `sizeof(float)` is no different to making it equal to `sizeof(float)`. It is still moving to the byte after `f`.

pakhontas, 2015-09-22:
In this case comment to row is wrong
It's not move.
in this description, I think more accurately comment:
https://www.arduino.cc/en/Reference/EEPROMPut  (eeAddress += sizeof(float);)

Chris--A, 2015-09-22:
Yes, the address 'pointer' is moving to the location specified by the comment. Whether its explicit or compounded is irrelevant.

pakhontas, 2015-09-22:
if first value of eeAddress set to not zero.
code in Get() becomes wrong.
If you think that just fine, of course, that is your right.
but I think that + = is understandable. thank.

Chris--A, 2015-09-22:
Of course, there is an issue if the previous address is not `0`. I guess for a newbie point of view, its always best to follow 'KISS' principles. I will update the bundled examples.

pakhontas, 2015-09-22:
Yes, it would be better for LEARNING section.