arduino/Arduino #3844: Arduino Leonardo Problem with windows 10 when OS reboot

Snower4344, 2015-09-23:
Arduino Leonardo
OS: Windows 10 
IDE: 1.6.5

I have install arduino driver correctly and it's work fine under Win10 

It's show Arduino Leonardo(COM1) in device manager's COM port category.

After the OS restart, the OS can't recognized this arduino device
It's in device manager's other devices category  and can't communicate with it.

I re-install driver, but it's still not working.
I must click the Reset button on the board or unplug and plug the USB connector every time i reboot OS on my pc.
 
I think this is not smart and cause everyone's trouble.

By the way, It's all work fine on Win7 & Win 8.

Anyone or Official arduino who can help to save this driver's problem on windows 10 ?  

Thanks for your help~

ffissore, 2015-09-23:
I don't think it's a driver problem. If it was, you Leonardo wouldn't work at all, reboot/replug or not. This looks like a W10 specific issue. Try searching the [forum](http://forum.arduino.cc/).

closed 2015-09-23 by ffissore

Snower4344, 2015-09-23:
I think Arduino also has responsibility on this problem. 
If arduino want to popularize their SW & HW to the world, they should test their compatibility on these OS.  
This problem can be reappear on Win10.

facchinm, 2015-09-23:
Hi @Snower4344 ,
I've just tested on Win10 and the Leonardo is correctly recognized after a reboot.
Anyway, I'm using the 1.6.6 nightly and default drivers (not Arduino one). 

Next AVR core version will include some fixes to USB handling, maybe you can try to test it by flashing a sketch with nightly IDE before rebooting.