arduino/Arduino #3845: Ethernet library seem broken in nightly builds

fallberg, 2015-09-23:
```
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp: In member function 'int DNSClient::inet_aton(const char*, IPAddress&)':
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:65:13: error: 'address' was not declared in this scope
     while (*address)
             ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp: At global scope:
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:58:5: warning: unused parameter 'addr' [-Wunused-parameter]
 int DNSClient::inet_aton(const char* addr, IPAddress& result)
     ^
In file included from /opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:7:0:
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp: In member function 'uint16_t DNSClient::ProcessResponse(uint16_t, IPAddress&)':
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:267:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t header_flags = htons(*((uint16_t*)&header[2]));
                                                          ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:4:22: note: in definition of macro 'htons'
 #define htons(x) ( ((x)<< 8 & 0xFF00) | \
                      ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:267:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t header_flags = htons(*((uint16_t*)&header[2]));
                                                          ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:5:22: note: in definition of macro 'htons'
                    ((x)>> 8 & 0x00FF) )
                      ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:269:50: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     if ( ( iRequestId != (*((uint16_t*)&header[0])) ) ||
                                                  ^
In file included from /opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:7:0:
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:286:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t answerCount = htons(*((uint16_t*)&header[6]));
                                                         ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:4:22: note: in definition of macro 'htons'
 #define htons(x) ( ((x)<< 8 & 0xFF00) | \
                      ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:286:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     uint16_t answerCount = htons(*((uint16_t*)&header[6]));
                                                         ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:5:22: note: in definition of macro 'htons'
                    ((x)>> 8 & 0x00FF) )
                      ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:295:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     for (uint16_t i =0; i < htons(*((uint16_t*)&header[4])); i++)
                                                          ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:4:22: note: in definition of macro 'htons'
 #define htons(x) ( ((x)<< 8 & 0xFF00) | \
                      ^
/opt/arduino-nightly/libraries/Ethernet/src/Dns.cpp:295:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     for (uint16_t i =0; i < htons(*((uint16_t*)&header[4])); i++)
                                                          ^
/opt/arduino-nightly/libraries/Ethernet/src/utility/util.h:5:22: note: in definition of macro 'htons'
                    ((x)>> 8 & 0x00FF) )
                      ^
exit status 1
Error compiling.
```

cmaglie, 2015-09-23:
Fixed. Thank you for the bug report!

closed 2015-09-23 by cmaglie

fallberg, 2015-09-23:
Thank you for fixing.

commit 0083d752 (2015-09-23): Fixed typo in Ethernet lib

Fix #3845