arduino/Arduino #3851: Weirdness with some platforms not being visible

matthijskooijman, 2015-09-23:
I have a bit of a peculiar setup, apparently, which seems to confuses the Arduino IDE. My setup is that I have a symlink `Sketchbook/hardware/arduino/avr` pointing to `arduino_git_checkout/hardware/arduino/avr`. The reasoning here is that, while working on the Arduino AVR core files, I can test them by hitting  recompile in the IDE, instead of also having to recompile the IDE itself to copy the right files into `build/linux/work` where they are normally loaded from.

So far, this setup has worked great (and, for the AVR core, it still does). However, I recently tried to install the SAMD core through the boards manager. It is now installed, and listed as such in the board manager, but its boards are not shown in the menu.

I did a bit of digging, and [found out that BaseNoGui](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L606-L608):
 1. Loads any packages/platforms from the builtin hardware directory.
 2. Loads packages/platforms from the `ContributionsIndexer` (e.g. any platforms and versions listed in the package JSON files).
 3. Loads packages/platforms from the sketchbook hardware directory.
 
However, in each step, it completely overwrites the full (previous) package. Since the `ContributionsIndexer` actually looks at the builtin hardware folder (and possibly the sketch folder as well?), that won't run into problems (as long as the versions are all listed in the JSON). However, the last step overwrites the "arduino" package with a package containing just the "avr" platform from my sketchbook directory.

While writing this, I realized that I should probably just clone the samd repository too and add a similar symlink to it, which should probably fix my problem. However, I do wonder if this handling can somehow be improved. While reading the code, I actually wondered if we really need all this special handling of "packages" in the first place? Perhaps we can just use a flat list of platforms only, using the package name as a name prefix to qualify the platform names, but without having any Java objects for them?

matthijskooijman, 2015-09-23:
I just tried adding the symlink to samd, which seems to work (it even picks up the tools needed from ~/.arduino15/packages, nice).

ffissore, 2015-09-25:
Uhm I didn't know but I'm not surprised. At least, it's all solved in arduino-builder, which merges all definitions it finds instead of replacing them. I'll fix that on the IDE side too

ffissore, 2015-09-25:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-09-25 by ffissore

matthijskooijman, 2015-09-25:
Looks good, thanks!

commit b82aebd9 (2015-09-25): If a package is already defined, don't replace it with the one found (for example) in sketchbook/hardware, but merge their contents. Fixes #3851