arduino/Arduino #3856: Arduino crashs on invalid Category

NicoHood, 2015-09-24:
```
2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] Invalid library found in /Arduino/build/linux/work/libraries/Firmata-2.4.4: /Arduino/build/linux/work/libraries/Firmata-2.4.4
     [exec] WARNING: Category 'Testing' in library libinclude is not valid. Setting to 'Uncategorized'
     [exec] java.lang.NoSuchMethodError: org.fife.ui.rtextarea.RTextScrollPane.<init>(Lorg/fife/ui/rtextarea/RTextArea;Z)V
     [exec] 	at processing.app.Editor.<init>(Editor.java:289)
     [exec] 	at processing.app.Base.handleOpen(Base.java:875)
     [exec] 	at processing.app.Base.handleOpen(Base.java:860)
     [exec] 	at processing.app.Base.handleOpen(Base.java:856)
     [exec] 	at processing.app.Base.handleNew(Base.java:751)
     [exec] 	at processing.app.Base.<init>(Base.java:455)
     [exec] 	at processing.app.Base.guardedMain(Base.java:219)
     [exec] 	at processing.app.Base.main(Base.java:135)
```

https://github.com/ricklon/libinclude

This error occurred in build https://github.com/arduino/Arduino/tree/fc8f84f06b919b110c38bc69590e065cec0f379c (thats the version I used, I dont know when the error started)

The error was not there at the time before the arduino-builder.

NicoHood, 2015-09-24:
Deleting/renaming the linux/work folder solved the problem. With a fresh build it worked, but the error is still listed in the command line. I kept the old /work folder if you want to debug it.

ffissore, 2015-09-24:
I guess you just had some stale jar files in your work folder, since cleaning it solved. as for firmata-2.4.4 folder, I guess it's a problem with unzip. I don't have it because I use the my java IDE to start the arduino IDE. @cmaglie is having the same issue. How about removing "build" as a dependency of "run" in build.xml?

cmaglie, 2015-09-24:
```
     [exec] java.lang.NoSuchMethodError: org.fife.ui.rtextarea.RTextScrollPane.<init>(Lorg/fife/ui/rtextarea/RTextArea;Z)V
```

this one looks like a leftover .class file that needs to be recompiled but that ant didn't detect it as outdated. It's a bit rare but happens, in this case an `ant clean` solves the issue.

> @cmaglie is having the same issue

Yes, the Firmata library is incorrectly expanded the second time you do `ant run`.
I'd prefer to not remove the dependency "run"->"build", because I use `ant run` as a synonym for `ant build run` and I guess that many people do the same.

matthijskooijman, 2015-09-24:
I've seen this error as well yesterday, `ant clean && ant` fixed this for me.

closed 2015-09-25 by ffissore