arduino/Arduino #3857: Add examples for hardware folders

NicoHood, 2015-09-24:
It would be nice it hardware folders also could contain example folder which are recognized by the IDE. This would also get rid of the warning errors:
`[exec] Could not find boards.txt in /Arduino/sketchbook/hardware/HoodLoader2/examples. Is it pre-1.5?`

ffissore, 2015-09-24:
That warning is about an invalid folder structure, indeed pre 1.5. It should be hardare/Hoodloader2/avr

NicoHood, 2015-09-24:
But where to put the examples? Inside avr?

ffissore, 2015-09-25:
ATM you can't deliver examples for cores. You can only add them to libraries, whether they are bundled with a core or not

NicoHood, 2015-09-25:
Exactly. And I want to add core examples, since my HoodLoader2 has some extra features/explanation/examples to show what is different.
It would be nice to at least avoid the warning first and then add a proper IDE menu when you have time for this.

ffissore, 2015-09-25:
In order to avoid the warning, move that examples folder away. It won't be the right place anyway, since at that level you specify the arch

NicoHood, 2015-09-28:
A behavior like this would be appreciated (without workaround via library):
https://github.com/NicoHood/HoodLoader2/pull/23

sandeepmistry, 2016-03-09:
Marking this a duplicate of #3297.

closed 2016-03-09 by sandeepmistry