arduino/Arduino #3858: [DOC] https://www.arduino.cc/en/Hacking/Bootloader page is very outdated

q2dg, 2015-09-24:
"Not using a bootloader", "Burning the bootloader", "Version of the bootloader" and "How does it work?" sections of this page should be entirely rewritten to reflect the behaviour of "new"boards like ATmega32U4-based, Due and Zero (at least).

Moreover, "bootloader source code" link (at the end of the text) should be updated too.

Thanks

agdl, 2015-09-24:
ok, for now link updated ;)

q2dg, 2015-09-24:
Shouldn't it target to https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/bootloaders (AVR bootloaders) and https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero (Zero bootloader) ??? (If I'm not wrong, Due bootloader is from Atmel's default, if so, this fact should be noticed too)

agdl, 2015-12-17:
@q2dg I'm assigning this issue to @facchinm because he and @cmaglie are working on the new Arduino boot-loaders so they can update the pages in the proper way with the latest infos.

facchinm, 2019-01-02:
@fragmuffin link fixed, thanks for the heads up. The original issue reported other section to be fixed which still need to be updated so I'll leave it open.