arduino/Arduino #3859: Closing the Serial Port freezes IDE [linux]

NicoHood, 2015-09-24:
Open the IDE, open a Serial Port, close it and it will freeze

```
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchFieldError: textArea
	at processing.app.SerialMonitor.close(SerialMonitor.java:107)
	at processing.app.AbstractMonitor$1.windowClosing(AbstractMonitor.java:34)
	at java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:349)
	at java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:349)
	at java.awt.Window.processWindowEvent(Window.java:2058)
	at javax.swing.JFrame.processWindowEvent(JFrame.java:305)
	at java.awt.Window.processEvent(Window.java:2017)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

ffissore, 2015-09-24:
Same issue as #3856. If the error was true, the IDE wouldn't even compile. `ant clean build` to the rescue

closed 2015-09-24 by ffissore

NicoHood, 2015-09-24:
Hm weird. You were right. I better to `ant clean build run` now.