arduino/Arduino #3867: Path issue in linux version

Adcade2, 2015-09-25:
The linux 64bits version of arduino, has several errors concernig paths to resources, as many of them lead to **/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/**  (there is no user called jenkkins on my machine).

Using the comand `grep -r jenkins *` in the arduino folder returns:
`Coincidencia en el archivo binario arduino-builder
hardware/tools/avr/lib/pkgconfig/libusb-1.0.pc:prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir
hardware/tools/avr/lib/pkgconfig/libusb.pc:prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir
hardware/tools/avr/lib/gcc/avr/4.8.1/plugin/include/configargs.h:static const char configuration_arguments[] = "../gcc-4.8.1/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir --enable-long-long --disable-nls --disable-checking --disable-libssp --disable-libada --disable-shared --enable-lto --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr";
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/tiny-stack/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/tiny-stack/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega7/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega7/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega2/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega2/libgcov.a
hardware/tools/avr/lib/gcc/avr/4.8.1/install-tools/mkheaders.conf:SYSTEM_HEADER_DIR="/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/avr/sys-include"
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega5/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega5/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr35/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr35/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega6/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega6/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr25/tiny-stack/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr25/tiny-stack/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr25/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr25/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr6/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr6/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr5/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr5/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega4/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrxmega4/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr4/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr4/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrtiny/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avrtiny/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr51/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr51/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr31/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr31/libgcov.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr3/libgcc.a
Coincidencia en el archivo binario hardware/tools/avr/lib/gcc/avr/4.8.1/avr3/libgcov.a
hardware/tools/avr/lib/libusb.la:dependency_libs=' -L/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/lib /home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/lib/libusb-1.0.la -lrt'
hardware/tools/avr/lib/libusb.la:libdir='/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/lib'
hardware/tools/avr/lib/libusb-1.0.la:libdir='/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/lib'
Coincidencia en el archivo binario hardware/tools/avr/lib/libusb-0.1.so.4.4.4
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-as
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-strings
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-g++
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-addr2line
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gdb
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-cpp
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gcc-ranlib
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-ar
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gcc-4.8.1
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gprof
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-run
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gcc
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-strip
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-ld.bfd
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-size
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-ld
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-c++
Coincidencia en el archivo binario hardware/tools/avr/bin/avrdude_bin
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-objdump
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-nm
hardware/tools/avr/bin/avr-man:prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir
hardware/tools/avr/bin/avr-man:datadir=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/share
hardware/tools/avr/bin/avr-man:exec man -M /home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/share/doc/avr-libc-1.8.0svn/man "$@"
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-ranlib
hardware/tools/avr/bin/libusb-config:prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gcc-nm
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-c++filt
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-gcc-ar
Coincidencia en el archivo binario hardware/tools/avr/bin/avr-objcopy
hardware/tools/avr/x86_64-unknown-linux-gnu/avr/lib/libbfd.la:libdir='/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/x86_64-unknown-linux-gnu/avr/lib'
hardware/tools/avr/x86_64-unknown-linux-gnu/avr/lib/libopcodes.la:libdir='/homejenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/x86_64-unknown-linux-gnu/avr/lib'
Coincidencia en el archivo binario hardware/tools/avr/x86_64-unknown-linux-gnu/avr/lib/libbfd.a
hardware/tools/avr/libexec/gcc/avr/4.8.1/liblto_plugin.la:libdir='/home/jenkinsjenkins/jobs/toolchain-avr-linux64/workspace/objdir/libexec/gcc/avr/4.8.1'
hardware/tools/avr/libexec/gcc/avr/4.8.1/install-tools/mkheaders:  prefix=/homejenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir
Coincidencia en el archivo binario hardware/tools/avr/libexec/gcc/avr/4.8.1/cc1
Coincidencia en el archivo binario hardware/tools/avr/libexec/gcc/avr/4.8.1/cc1plus
Coincidencia en el archivo binario hardware/tools/avr/libexec/gcc/avr/4.8.1/lto1
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/c++
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/nm
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/gcc
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/ld
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/strip
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/objdump
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/ranlib
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/ar
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/g++
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/as
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/objcopy
Coincidencia en el archivo binario hardware/tools/avr/avr/bin/ld.bfd`

I tried with the last stable version (Arduino 1.6.5-r5) as wel with nightly builds, and the issue is the same.

How can I fix it without modifying one by one the files (and binaries) that contain this path?

matthijskooijman, 2015-09-26:
Can you include the actual errors you ar seeing that you want to see fixed? The jenkins path is probably included in a lot of those binaries, since that is the path in which they were built (which is included in the meta information as well, I believe).

Adcade2, 2015-09-26:
When I try to talk directly to avrdude with `~/.arduino-1.6.5-r5/hardware/tools/avr/bin/avrdude -vv`, I get this error:

```
avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
        Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
        Copyright (c) 2007-2009 Joerg Wunsch

        System wide configuration file is "/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/etc/avrdude.conf"
avrdude: can't open config file "/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir/etc/avrdude.conf"
```

The avrdude.conf file is located at `/etc/avrdude.conf` and in `~/.arduino-1.6.5-r5/hardware/tools/avr/etc/avrdude.conf`

matthijskooijman, 2015-09-27:
The IDE passes an explicit -C option to avrdude to locatie the .conf file. I do not think that changing the default path in the avrdude binary would really help here: You could point it to /etc/, but that file is likely for another avrdude version and will not work (potentially leading to hard-to-debug problems) and the path to ~/.arduino-1.6.5-r5/hardware/tools/avr/etc/avrdude.conf depends on where you install Arduino, so cannot be hardcoded (and I suspect avrdude does not allow relative paths).

So I think the solution is to pass -C to avrdude, and nothing in the IDE / toolchain can be / needs to be changed. I'm closing this issue, feel free to leave a comment if you think otherwise (then we can reopen).

closed 2015-09-27 by matthijskooijman

HP-Sparks, 2016-06-21:
Hi,
I'm facing a similar issue bringing me there with Arduino IDE 1.6.9 64-bit under Linux Mint 17.3 Cinnamon 64-bit. (Ubuntu-based/derived distribution)
What would be a decent workaround until this get fixed ?

matthijskooijman, 2016-06-21:
@HP-Sparks, best to open up a new issue. Make sure to include the full verbose compilation and upload output, and describe the actual problem you have (can't really offer a workaround or fix if we don't know what problem you have). For the issue described above, I believe the problem was in the platform.txt of a third-party core, but for lack of a response (or full output), I can't tell for sure.

HP-Sparks, 2016-06-21:
@matthijskooijman, Thank you for your reply.
I also thought better to open a new record for my personal problem.
You may find it there: [https://github.com/arduino/Arduino/issues/5059](url)
I'm Not sure if I provided enough information in the initial post. Only what the IDE shown me.
I'm going to enable the "verbose" mode to try to learn more.
Thanks again !

Regards,
  HP_