arduino/Arduino #3869: Wrong Bootloader not present warning

NicoHood, 2015-09-26:
If I compile a sketch for my custom HoodLoader2 core I now get a warning that the .hex file is missing. However a ISP programming still works. The IDE looks at the internal path, not the hardwarefolder where the HoodLoader2 is installed.
https://github.com/NicoHood/HoodLoader2/tree/dev_205

Also I suggest to somehow access the default bootloaders. For example I added a copy of the Uno/Mega board to add the Serial 1200 baud touch feature. But I still want to use the optiboot bootloader. At the moment I have to put a copy of this hexfile in my repository which I do not prefer since this can be avoided with `arduino:optiboothexfilepath`.

IDE Warning (when hitting compile):
```
/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -ide-version=10606 -build-path "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled1542786869.tmp/sketch_sep26a/sketch_sep26a.ino"
/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware "/Arduino/build/linux/work/hardware" -hardware "/home/linuxuser/.arduino15/packages" -hardware "/sketchbook/hardware" -tools "/Arduino/build/linux/work/tools-builder" -tools "/Arduino/build/linux/work/hardware/tools/avr" -tools "/home/linuxuser/.arduino15/packages" -libraries "/sketchbook/libraries" -libraries "/Arduino/build/linux/work/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -ide-version=10606 -build-path "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/tmp/untitled1542786869.tmp/sketch_sep26a/sketch_sep26a.ino"
WARNING: Spurious .git folder in 'AnalogTouch' library
WARNING: Spurious .git folder in 'HID-Project' library
WARNING: Spurious .git folder in 'IRLremote' library
WARNING: Spurious .git folder in 'MSGEQ7' library
WARNING: Spurious .git folder in 'Nintendo' library
WARNING: Spurious .git folder in 'PinChangeInterrupt' library
WARNING: Spurious .git folder in 'RCLSwitch' library
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16  "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16       "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/tools-builder/ctags/5.8-patched/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/preproc/ctags_target.cpp"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/sketch/sketch_sep26a.ino.cpp.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/wiring_pulse.S.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/WInterrupts.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/hooks.c" -o "/tmp/buildac6989e7f7f14d9950cca6539075540b.tmp/core/hooks.c.o"
"/Arduino/build/linux/work/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/sketchbook/hardware/HoodLoader2/avr/variants/HoodLoader2" "/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/wiring.c" -o "/tmp/buildac [... cut ...]
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

per1234, 2015-09-26:
> I now get a warning that the .hex file is missing.

See https://github.com/arduino/Arduino/issues/3831
> I suggest to somehow access the default bootloaders.

https://github.com/arduino/Arduino/issues/2656 (and I agree, this is a very necessary feature)

ffissore, 2015-09-28:
This issue was moved to arduino/arduino-builder#20

closed 2015-09-28 by ffissore

NicoHood, 2015-10-02:
I'd like to reopen this in a similar context.
I think the check should look if a bootloader file is present. However if you use an ISP it tries to upload everything and in the finial hex file reading it fails if the bootloader doesnt exist. The program should catch this before starting avrdude.

Used this build:
https://github.com/arduino/Arduino/pull/3896#issuecomment-144994262

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e9587
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/home/linuxuser/Documents/Arduino/sketchbook/hardware/pololu/avr/bootloaders/caterina/Leonardo-prod-firmware-2012-12-10"
avrdude: can't open input file /home/linuxuser/Documents/Arduino/sketchbook/hardware/pololu/avr/bootloaders/caterina/Leonardo-prod-firmware-2012-12-10: No such file or directory
avrdude: read from file '/home/linuxuser/Documents/Arduino/sketchbook/hardware/pololu/avr/bootloaders/caterina/Leonardo-prod-firmware-2012-12-10' failed

avrdude done.  Thank you.

Error while burning bootloader.
```