arduino/Arduino #3875: Information inconsistencies between information showed in Library Manager and in web doc

q2dg, 2015-09-27:
AudioZero library isn't listed in Library Manager!!!

    RTCZero library is called "RTC" on documentation (https://www.arduino.cc/en/Reference/RTC)

    USBHost library is described as "Due only" in Library Manager but in documentation is said it's compatible with Zero too.
 
    EEPROM library is described as "All boards except Due" but it should say "All boards except Due and Zero
 
     https://www.arduino.cc/en/Reference/HID page doesn't exit

agdl, 2015-09-28:
Ho @q2dg 
* AudioZero has been added to library manager as you pointed out;
* this can be asked to @facchinm 
* you are right, but this is because in the future we want to have an homogeneous RTC library for RTCs in general. In this way we have something pre-build
* added Zero to the README;
* added Zero to the README;

q2dg, 2015-09-28:
Thanks a lot ! I'll close this issue and will open another one remembering to Martino the missing https://www.arduino.cc/en/Reference/HID page , then.

closed 2015-09-28 by q2dg