arduino/Arduino #3877: Problem in build process when library include another library

descampsa, 2015-09-27:
I think i've found a quirk in the build process, described below:

If you compile a library A that  include header of another library B, without including library B in the skecth, compilation fails, which is normal (according to your build process works, at least).
But, if you include library B in the skecth, compile, and then remove the include of library B, it will still compile fine (link will fail if you actually use library B, but that's not the point).  This is true even if library A is modified and recompiled.

Tested with version 1.6.5

descampsa, 2015-09-27:
Hmm, after trying it again, it seems that my last sentence is not true : if library A is modified and recompiled, the compilation fails, as expected. 
So it the 'problem' is just that libraries are not recompiled when sketch include list is changed. 
This can be a problem because the include list can have an impact on the library compilation.

I agree that it is a rather rare case and that the solution (recompilation everytime include list is changed) is maybe not better than the problem.

Up to you to decide

ffissore, 2015-09-28:
This is a known limitation of the IDE. There is no library to library dependency resolution. Some work has been done with upcoming 1.6.6 and its new `arduino-builder` tools. See https://github.com/arduino/Arduino/pull/2792#issuecomment-141485856 and https://github.com/arduino/arduino-builder/issues/12.

matthijskooijman, 2015-09-28:
@ffisore, I actually think the behaviour that this issue described is not that dependencies between libraries are not supported, but it's that if the sketch is modified to include different libraries, which changes the environment (include path) for all libraries, those libraries are not recompiled, potentially causing unexpected results. In practice, this is a corner case that will mostly change behaviour in cases that are already broken I think, but it might be nice to fix it anyway.

A quick fix is to add the list of libraries, or the include path, to the build properties JSON that used to decide wether to recompile all. A better fix (now that library-to-library dependencies are supported) would be to give each library its own include path (based on detecting the dependencies of just that library), instead of building one big include path based on the joint dependencies of all libraries. This could even prevent conflicts in some cases (consider two libraries each depending on one of two other libraries that provide the same (secondary) header file, which will now certainly cause a conflict).

descampsa, 2015-09-28:
@matthijskooijman , your interpretation is correct.
I actually had not much hope that this issue would be fixed, because i didn't know that library-to-library dependency was now officially supported. That is good news!
Personnally, i think your second solution would be the most consistent behaviour, even if not perfectly backward compatible.

Thanks

bperrybap, 2015-09-28:
So what does " library-to-library dependencies are supported" really mean?
In my mind it would mean that if a library includes a header for another library then the IDE would be smart enough to add that libraries directory to the include path, build it, and link it in. But I believe that the IDE is not doing this, so what does that statement really mean?

matthijskooijman, 2015-09-28:
> In my mind it would mean that if a library includes a header for another library then the IDE would be smart enough to add that library's directory to the include path, build it, and link it in. 

The new arduino-builder in git master / hourly builds does exactly this (though it isn't perfect yet, should be perfected soon).

matthijskooijman, 2015-09-28:
> Personnally, i think your second solution would be the most consistent behaviour, even if not perfectly backward compatible.

Well, since library-to-library dependencies haven't been supported in any official release yet, changing this behaviour now won't cause any compatibility issues (which is an extra reason to tackle this right away, instead of waiting).

@descampsa, could you perhaps try the hourly build to confirm this issue is also present there, and if so open an issue in the arduino-builder repository (which is where this should really be fixed)?

PaulStoffregen, 2015-09-28:
@bperrybap - Are you running the latest hourly build?  (or any of them from the last several days)  Much has changed recently with the new builder program.

bperrybap, 2015-09-28:
Paul,
I haven't run anything newer than 1.6.5. But that sounds great. It sounds like one of the best features/capabilities added to the IDE in quite some time. It will solve many issues. I'll have to check it out.

descampsa, 2015-09-29:
> Well, since library-to-library dependencies haven't been supported in any official release yet, changing this behaviour now won't cause any compatibility issues (which is an extra reason to tackle this right away, instead of waiting).

Yeah, i meant it could break some code that use unsuported features, but you're right, it shoud'nt break any officially supported feature.

> @descampsa, could you perhaps try the hourly build to confirm this issue is also present there, and if so open an issue in the arduino-builder repository (which is where this should really be fixed)?

Ok, i will do that.

ffissore, 2015-09-30:
Also asking here: can you provide an example sketch + libraries, even fake libraries?

ffissore, 2015-10-07:
Closing as tracked at https://github.com/arduino/arduino-builder/issues/22

closed 2015-10-07 by ffissore