arduino/Arduino #3879: Support 31250 baud in the Serial Monitor

sdsea, 2015-09-28:
I'd like the Serial Monitor window to support a baud rate of 31250.  I use Arduinos to develop MIDI projects.  MIDI also uses serial communication, but at a speed of 31250.  While debugging, I have to switch the output to the serial monitor to debug what is outputted.  When I do this, I also have to change the serial speed.  It would be nice if the IDE supported a serial rate of 31250.  This would be one less thing I'd have to change when going back and forth.

Thanks,

Jeff

matthijskooijman, 2015-09-28:
I suspect this might be problematic on Linux, where non-standard baudrates (such I think this one is) are a bit tricky to support (see https://github.com/arduino/Arduino/issues/3351#issuecomment-113873045). I guess this will be wontfix - you'll have to use a different terminal emulator that knows how to properly set custom baudrates.

Chris--A, 2015-09-28:
If this is a valuable speed to have (used by many), it should be fine as @cmaglie added a check for incompatible baud rates on linux, so it will at least give a useful error (When I added 74880 for the ESP8266).

https://github.com/arduino/Arduino/pull/3420

NicoHood, 2015-09-28:
Oh I did not know of the last PR chris linked. I really like to see some more baud rates and also linux support for them. Other terminal program can do this too. Why does it work with windows and not with linux. Thats embarrassing. At least a warning would be nice, but the option for windows users to enable this feature.

Also a table of supported linux baud rates would be nice. There are some online but not all work 100% especially for a cdc bootloader or so. Even very simple baud rates like 300 caused problems for me.

matthijskooijman, 2015-09-28:
> I really like to see some more baud rates and also linux support for them. Other terminal program can do this too. Why does it work with windows and not with linux. Thats embarrassing. At least a warning would be nice, but the option for windows users to enable this feature.

I think I thoroughly explained the situation in the comment linked above. The PR Chris linked will only cause an error message in Linux when trying to use an unsupported baudrate. I guess that means that adding baudrates that are not supported on Linux might be feasible now, to at least get them working for Windows and perhaps Mac users (not sure how the custom baud rate situation on those platforms is).

If you really want to get more baudrates supported on Linux, the only way is to fix JSSC to support the right APIs to configure them.

> Also a table of supported linux baud rates would be nice. There are some online but not all work 100% especially for a cdc bootloader or so. Even very simple baud rates like 300 caused problems for me.

Feel free to test and compile a list, but I think that it can be somewhat system-dependent. The JSSC limitations means that arbitrary baudrates (say 12345 bps) are not supported, only well-known baudrates. If 300bps isn't working, this is probably a bug in another part of the system. Ideally, al bugs would be fixed and JSSC is extended, but each of these are different issues and should be tracked and investigated as such. I agree that it would be nice to fix these, but I suspect it's not a priority for the Arduino team.

sdsea, 2015-09-29:
@matthijskooijman: If this is not possible on Linux, but is feasible on Windows and OSX, how about a compromise then and support it under Windows and OSX, but not Linux?  Maybe use an #ifdef to target these specific platforms?  Not ideal, I know, but would still be nice to have.

@Chris--A: To those doing MIDI development this is valuable speed.  MIDI has been around for 30 years now (amazing!).  It's a serial communication standard, and the speed is always 31250 baud.

I hope this can get in, and thanks for the consideration,

Jeff

matthijskooijman, 2016-01-17:
I'm closing this as a duplicate of #4343, which somewhat more generically addresses this issue.

closed 2016-01-17 by matthijskooijman

rolfmeurer, 2018-09-30:
I'm using HTerm @ 28800 8n1 to debug single events, that works for 31250 BAUD coming from ARDUION, but not for serial streams, as I said, just for single events. usualy I have MIDIOX running on a 2nd PC connected with a MIDI interface to the serial ARDUINO output. This is really a comfortable tool.