arduino/Arduino #3880: [DOC] Missing https://www.arduino.cc/en/Reference/HID page

q2dg, 2015-09-28:
When HID library is selected in Library Manager, its "See more" link points to this address, which doesn't exist. 

*@facchinm On the other hand, what's that library??...Is it based on https://github.com/NicoHood/HID/tree/dev_2_4?? Does it deprecate Keyboard and Mouse libraries?.... Thanks.*

NicoHood, 2015-09-28:
I tried to upgrade the HID library with some additional features. If its partly integrated (at least the pluggable HID key features) is under discussion.

An HID documentation would be nice, a link to my library would be cool.

facchinm, 2015-09-29:
Hi @q2dg , the HID library is a porting of existing code which lived inside the core, Nico's one is a (very good) rewrite/expansion of the original code.
A sketch compiles without the inclusion of `HID.h` thanks to @ffissore 's work on recursive library detection, which is still in beta so the documentation has not been updated.
We will stabilize the documentation once the library APIs will be stable (quite soon I hope) :wink:

q2dg, 2015-09-29:
Ok, thanks a lot for your explanation. I'll close this issue, then, because I infer you are already working on it and you don't need this thread to be open.

I have a last question, though: is HID library intended to be used by a sketch developer or it's like a "hidden" library designed to support Keyboard/Mouse/... ones? I ask this because HID library appears as another regular library inside the list of standard libraries shown by the IDE, so I don't know if this fact could be a little confusing. But maybe I'm completely wrong.

facchinm, 2015-09-29:
Well, you could use it directly in a sketch but without defining a proper descriptor (and thus implementing a real device) it is pretty useless.
I'm reopening the issue as a reminder, will close it when the docs are ok!

closed 2015-11-18 by facchinm