arduino/Arduino #3885: DUE as i2c slave: i2c connx  to a master don't work - for the Mega code works fine

VogonJeltz, 2015-09-28:
I am running my DUE as an i2c slave to communicate to a i2c master (bitbanged i2c, it's a EV3): 
the Due connx  to the  master don't work - to my Uno and  Mega my code works fine nevertheless.
The code establishes a communication protocol to exchange data packages (arrays) between Arduino and EV3.

There are (at least) 2 possible 

(1)
For the Mega I compellingly had to disable the internal pullups (patch in twi.c - ( for the Uno this was not necessary as there aren't internal pullups at all).
~~~
//activate internal pullups for twi.
//digitalWrite(SDA, 1);
//digitalWrite(SCL, 1);
~~~
for the Due I dont know if the patch works also for the DUE because twi.c is located in an avr subdir
As external pullups 2x 47k have to be used (actually 30-80 are fine, tested by Uno and Mega), 

(2)
there were issue reports about the i2c implementation on the DUE.
I tried both the old version coming with the 1.6.5 release and the pasch  copied to the Wire subdir.
Again I don't know neither if it affects the DUE as the Wire subdir again is in a avr subdir, nor if the patch generally is suited to fix the error in principle.

For the Due I both tested Pullups to 3.3V  and to 5V to play it safe. At  both levels it didn't work

Anyway, as the code works with Uno the Mega the issue is supposed to be located on the EV3 side.

First my question about (1): are there  internal pullups on the Due, and if they are: what about disabling.
Next question about Wire.cpp: are there known issues about ARM/DUE and how to fix them?

transmission source code (containing some irrelevant relicts from a former  UART protocol):
~~~
//  Arduino code to interface an Arduino to an EV3
//  Arduino as an I2C slave
//  compiles for MEGA and DUE, IDE 1.6.5
//  ver. 0.002a

#include  <SPI.h>
#include  <Wire.h>

//#include <ardustdio.h>

//=====================================================================================
// misc.  
//=====================================================================================

#define  clock()      millis()  

//=====================================================================================
// user interface:  button pad control pins
//=====================================================================================

#define  PIN_ESC    13
#define  PIN_UP     12
#define  PIN_OK     11
#define  PIN_DN      4 // instead opt.: 6
#define  PIN_LE      3 // instead opt.: 5
#define  PIN_RI      2

//=====================================================================================
// Digital Pins
//=====================================================================================

#define  SensorPTouch(pin) (!digitalRead(pin))    // btn press for _PULLUP Touch Pin (intern. pullup resistor)
#define  SensorTouch(pin)  ( digitalRead(pin))    // btn press for _PULLDOWN Touch Pin (ext. pulldown resistor)
#define  pbtn(pin)         (!digitalRead(pin))    // alias (_PULLUP Touch Pin)
 

//=====================================================================================

int16_t  btnpressed() {
   return ( pbtn(PIN_ESC)||pbtn(PIN_UP)||pbtn(PIN_OK)||pbtn(PIN_DN)||pbtn(PIN_LE)||pbtn(PIN_RI) );
}

//=====================================================================================

int16_t   getbtn() {
   int16_t  choice= -1;
   
   while (!  btnpressed() );  // wait until button pad pressed
   if( pbtn(PIN_ESC) ) choice = PIN_ESC;
   if( pbtn(PIN_UP) )  choice = PIN_UP;
   if( pbtn(PIN_OK) )  choice = PIN_OK;
   if( pbtn(PIN_DN) )  choice = PIN_DN;
   if( pbtn(PIN_LE) )  choice = PIN_LE;
   if( pbtn(PIN_RI) )  choice = PIN_RI;     
   while (  btnpressed() );   // wait until button pad released
   
   return choice;   
}

#define  SLAVE_ADDRESS 0x04
#define  MSGSIZE  30
byte     recvarray[MSGSIZE]; 
byte     sendarray[MSGSIZE];

int32_t  flag=0;

//=====================================================================================
//=====================================================================================
void setup() {
   char sbuf[128];   
   int32_t  i=0;
         
   // Serial terminal window
   i=115200;
   Serial.begin(i);  
   Serial.print("Serial started, baud="); Serial.println(i);
   
   // Wire (i2c)
   Wire.begin(SLAVE_ADDRESS);  //  start Arduino as a I2C slave, addr=0x04 (7-bit coded)
   Wire.onReceive(receiveData );
   Wire.onRequest(sendData );
   memset(sendarray, 0, sizeof(sendarray) );
   memset(recvarray, 0, sizeof(recvarray) );   
   Serial.println("I2C init: done.");
     
   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   

}

//=====================================================================================

uint8_t  calcchecksum(uint8_t array[]) {
  int32_t  sum=0;
  for(int i=4; i<MSGSIZE; ++i) sum+=(array[i]);
  return (sum & 0x00ff);
}

//=====================================================================================

void loop()
{  
  char sbuf[128];
   
  if(flag==1)
   {
     Serial.print(" sendarray[4]="); Serial.print(sendarray[4]);  // quick checking the flag bayte
     Serial.print(" recvarray[4]="); Serial.print(recvarray[4]);
     Serial.println(); Serial.println();
   
     // do something with the received data
     // and then do something to build the sendarray [4]...[31]
     sendarray[5] += 1;
     
     sendarray[0] = 0xff;
     sendarray[1] = calcchecksum(sendarray) ;
     sendarray[2] = SLAVE_ADDRESS; //ID check to handle several slaves
     sendarray[3] = flag; // 1= new data; 127=send again request
     //... feddisch!
     flag=0;     
   }
   else
   if(flag==127) {  // builds array with error flag 127
      memset(sendarray, 0, MSGSIZE);
      sendarray[3]=flag;
   }
  
   sprintf(sbuf, "Sendarr[4]=%4d,    [5]=%4d,    Recvarr[4]=%4d,    [5]=%4d", 
                  sendarray[4], sendarray[5], recvarray[4], recvarray[5]) ;
   Serial.println(sbuf);

   delay(2);  // short break for the cpu and the bus
}

//=====================================================================================

void receiveData(int byteCount) {
    int32_t i;
    byte val;
       
    while(Wire.available()<MSGSIZE) ; // wait for 30  bytes to complete
    i=0;  // init counter var
    while(Wire.available()&& (i<MSGSIZE) )         // read all bytes
    {
      val=Wire.read();
      recvarray[i++]=val;
    }
    // check for first 3 bytes and via calcchecksum() function
    if( (recvarray[0]==0xff) && (recvarray[1]==calcchecksum(recvarray)) && (SLAVE_ADDRESS==recvarray[2]) )    
    // if ok:
    flag=1;        // data ok
    else flag=127; // else handle receive error => flag =127  to send array back, request to send again
}

//=====================================================================================

void sendData(){
  // Wire.write writes data from a slave device in response to a request from a master
  Wire.write(sendarray, MSGSIZE); // send own 30 bytes back to master..
}

//=====================================================================================
~~~

VogonJeltz, 2015-09-30:
now are there buit-in i2c pullups on the DUE or not ?

VogonJeltz, 2015-09-30:
now are there buit-in i2c pullups on the DUE or not ? (it seems, there are some).

How to disable them by software (like on the Mega) ?

sandeepmistry, 2015-10-06:
@VogonJeltz there are built-in external pull-ups on the Due, they cannot be disable by software.

From the [schematic](https://www.arduino.cc/en/uploads/Main/arduino-Due-schematic.pdf):

[image]

VogonJeltz, 2015-10-06:
that's a pity - for the MEGA it's possible to disable the built-in pullups though.
Why not for the DUE?

sandeepmistry, 2015-11-04:
Duplicate of #4007

closed 2015-11-04 by sandeepmistry

VogonJeltz, 2015-11-04:
is this not the same as for the Mega?
[image]

sandeepmistry, 2015-11-11:
Please see the following comments from @mikaelpatel:  
* https://github.com/arduino/Arduino/issues/4007#issuecomment-150646533
* https://github.com/arduino/Arduino/issues/4007#issuecomment-150655807

As mentioned in comments linked above, from the [Mega schematic](https://www.arduino.cc/en/uploads/Main/arduino-mega2560_R3-sch.pdf):
[image]

VogonJeltz, 2015-11-13:
it's still not quite clear what the difference is between Due and Mega pullups.