arduino/Arduino #3889: Arduino Due watchdog support 1.6.5

kaddigat, 2015-09-29:
Watchdog does not work for arduino using IDE 1.6.5. However, according to the realease notes of 1.6.5 watchdog should already be implemented. 

https://www.arduino.cc/en/Main/ReleaseNotes

facchinm, 2015-09-29:
Hi @kaddigat , what version of SAM core is reported in Board Manager? 
Starting from IDE 1.6.2 non-AVR boards are handled externally so maybe you have an older version of the core while the IDE is updated. 

The watchdog has been added in this commit https://github.com/arduino/Arduino/pull/3111

kaddigat, 2015-09-29:
SAM core version is 1.6.4. But an update to 1.6.5 is not available. Is there a possibility to perform the update manually?

facchinm, 2015-09-29:
At the moment the only way to apply an almost `master` SAM core without manually patching the files is following these instructions (https://github.com/arduino/Arduino/pull/3524#issuecomment-136416350) while using a nightly version of the IDE. 

I'm really sorry for the inconvenience and I'll push for a fast release of the updated core.

kaddigat, 2015-09-29:
I tried to get the watchdog running with Arduino IDE 1.6.6 Hourly Build 2015/09/28 06:26. I put the latest SAM core (downloaded from https://mega.nz/#!R5JzSaYb!oRss5QcoL1Jv6pvULNzdcNulSQrmnDhtPKPo2F4qve4) into the sketchbook and chose the Arduino Due board from 'ARM boards PR 3524'. Everything seems to work fine. Then tested the watchdog with this basic sketch:

---------------------------------------
int counter = 0;

void setup() 
{
  watchdogEnable(1000);
  Serial.begin(9600);
}

void loop() 
{
  delay(500);
  Serial.println(counter);
  if(counter >= 10)
  { 
    while(true)
    {
      Serial.println("loop");
      delay(100);
    }
  }
  counter++;
}

------------------------------------

The compiler does not complain, but the Arduino does NOT restart und stays in the loop :(

facchinm, 2015-09-29:
Testing right now, I'll report the results here as soon as I can

facchinm, 2015-09-29:
Ok, found, there is a hole in the documentation about watchdog usage on DUE.
So, you need to define a non weak function
```C
watchdogSetup(void) {
// do what you want here
}
``` 
in your sketch to get things working right. 
Sorry for the inconvenience, I'll immediately start writing some docs about that!

kaddigat, 2015-09-29:
Thanks for the quick help! Everything is working fine now.

facchinm, 2015-09-29:
Glad it solved your issue!
IDE 1.6.6 is on its way with a lot of new features and it will bring the updated cores so stay tuned :wink:

closed 2015-09-29 by facchinm

q2dg, 2015-10-01:
Is there any chance to generalize this API to other boards (328-based, 2560-based, 32U4-based, Zero...) in order to homegenize this feature?
Thanks