arduino/Arduino #3891: Width Examples Menu and translation

pierrotm777, 2015-09-29:
Can you change the size of the Files/Examples menu ?
[image]
A good idea would be that the width size respect the length of the bigger title in the menu ! 

The good translate would be:
'Exemples des Lirairies Custom'

Thanks

ffissore, 2015-09-30:
Width should auto adjust. Try reopening the examples menu. I've see that behaviour before but I haven't dug into it that much

per1234, 2015-09-30:
I've noticed that the Board menu only auto adjusts to the width of items that are shown when the menu opens. If there are wider items that you have to scroll down to then they are cut off. To reproduce this:
- Add enough boards to make the menu scrollable.
- Change one of the board names or section names that will be off the screen to a long name
- Open the menu and scroll to make the long name appear, it will be cut off.
- Close the menu
- Reopen the menu, because it was scrolled down previously the long name is visible without scrolling and  the menu width is now auto adjusted to the width of the long name.
- Scroll the long name off the screen
- Close menu
- Open menu and scroll down to the long name, the name is cut off