arduino/Arduino #3895: Expand buildsystem with a few additions

NicoHood, 2015-09-29:
* A command to just run the ide without recompiling would be cool. "run" should be kept as it is, maybe use something like "start"?
* A command to copy the modified hardware folder. This is very annoying for testing. I currently need to copy my changes from linux/work into the real hardware folder (for a new commit). Also if I switch the branch and recompile the IDE all progress is lost. If i could simply edit the real (intended to edit) files and just copy them to my work directory (without hitting recompile) that'd be very nice.

ffissore, 2015-09-30:
@cmaglie we've talked about first @NicoHood request a few days ago. What's your opinion?
As for your second request, doesn't putting your hardware under `sketchbook/hardware" solve?

NicoHood, 2015-09-30:
Well but I still cannot switch branches via git fast enough. And I still need to copy them back for a commit. So there is no real improvement except the avoided data loss on rebuilding the ide.

ffissore, 2015-09-30:
What I usually do is to commit on some branch, switch to another, do some work, and switch back to previous branch. If previous branch needs the work done on the other, I rebase and git reset HEAD~1

ffissore, 2015-09-30:
ok, try now with `ant start`

matthijskooijman, 2015-09-30:
> A command to copy the modified hardware folder. This is very annoying for testing. I currently need to copy my changes from linux/work into the real hardware folder (for a new commit). Also if I [...]

I have a symlink from `sketchbook/hardware/arduino/avr` to `arduino_git_checkout/hardware/arduino/avr`, which pretty much solves this problem for me. This version overrides the version from `linux/work`, so you can edit in the git-controlled files directly, ignoring `linux/work` completely.

closed 2015-10-02 by ffissore

NicoHood, 2015-10-02:
Had no time to test this yet, but I guess it works as expected. Thanks for the addition :)
And thanks @matthijskooijman for the tip. will try that as well today.

NicoHood, 2015-10-03:
Works pretty well. Just added here:
https://github.com/arduino/Arduino/wiki/Building-Arduino