arduino/Arduino #3898: Can't use #if #else #endif

shellixyz, 2015-09-29:
It seems problematic to use these preprocessor directives with version 1.6.5. I discovered the issue when trying to upgrade from 10.6 to 1.6.5.

If I use #if #else #endif I get error messages because the output of the preprocessor is wrong.
e.g. firmware:45: error: expected unqualified-id before 'else'

If I remove them everything is fine. It is pretty useful and should be allowed. Just a bug ?

preprocessor output sample:
```c
#line 1 "firmware.ino"
#include <TimerOne.h>
#include "sample_buffer.h"
#include "pins.h"
#include "movement.h"
#include "adc.h"                                                                     
#include "Arduino.h"
void alloc_failure_trap();
void data_init();
void io_init();
void sync_proc_init();
void setup();
void sync_proc();
void sync_proc();
else if (ROTATING);
else if (ELEVATING);
void update_sampled_data();
void search_scan_debug();
void debug_temp();
void serial_send_temp(int8_t temp);
void serial_send_track_data();
void serial_send_light_intensity();
void loop();
else if (ROTATING);
#line 37
volatile shadow_samples_t shadow_samples;
volatile shadow_t shadow;
volatile sample_buffer_t head_temperature_samples;
volatile sample_buffer_t rot_track_error_samples, elv_track_error_samples;

volatile int8_t head_temperature;

ffissore, 2015-09-30:
This should be fixed with the new https://github.com/arduino/arduino-builder provided by the latest hourly build http://www.arduino.cc/en/Main/Software#hourly. Can you give it a try and report if your issue is still valid?

shellixyz, 2015-09-30:
I tested the last hourly build and it looks like it is now working. Nice, thank you.

closed 2015-09-30 by shellixyz