arduino/Arduino #3900: Java Exception closing Serial Monitor

Wackerbarth, 2015-09-30:
Build is MacOSX 10.10.5
IDE source c3c22ab in the past hour

Everything is fine until I connect to my device and then attempt to close the Serial Monitor window.

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchFieldError: textArea
	at processing.app.SerialMonitor.close(SerialMonitor.java:107)
	at processing.app.AbstractMonitor$1.windowClosing(AbstractMonitor.java:34)
	at java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:349)
	at java.awt.Window.processWindowEvent(Window.java:2058)
...

At this point, the IDE is "hung" and I have to Force Quit to kill it

ffissore, 2015-10-01:
Duplicate of #3859. Issue a full `ant clean build`. You have stale files from a previous build

closed 2015-10-01 by ffissore

Wackerbarth, 2015-10-01:
Thanks. The clean build fixed it.