arduino/Arduino #3901: Request to include TaskScheduler library into standard Arduino IDE

arkhipenko, 2015-09-30:
Hi,

Would you please consider including TaskScheduler library into Arduino IDE set of libraries?
The library is located here: https://github.com/arkhipenko/TaskScheduler.git

TaskScheduler is a light-weight cooperative multitasking library for arduino microcontrollers.
It enables developers to achieve pseudo multi-tasking quite easily, and have additional benefits. 
This library has been used by several people with generally very positive feedback, and extensively used by me.
The library is NOT pre-emptive (and as such requires cooperative programming to be used), does NOT use any of the timers (therefore does not affect PWM pins), and is very stable.

Regards,
Anatoli Arkhipenko

ffissore, 2015-10-01:
Added. Thanks!

closed 2015-10-01 by ffissore

PaulStoffregen, 2015-10-01:
How can this one be added, without  library.properties?

ffissore, 2015-10-01:
Thanks @PaulStoffregen, I missed the output of the job. While the lib repo has been added to the list, it didn't go into library manager because it's not valid. @arkhipenko please give a second read to https://github.com/arduino/arduino-builder/issues/15#issuecomment-144559284

ffissore, 2015-10-01:
Sorry I meant https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ (edited)

arkhipenko, 2015-10-01:
All fixed and committed back.